cantools/cantools #470: BU_BO_REL causes 'index out of range' error

CodenameReese, 2022-08-19:
Unfortunately I'm not very proficient with regards to the dbc format, but I noticed the following issue when trying to _generate_c_source_ form our CAN DB.

The following line causes an 'index out of range' error if the kind of a _BA_REL_ attribute is _BU_BO_REL_, since the sequence only has 6 elements:
https://github.com/cantools/cantools/blob/92f4c020df296402040eb84139bf7678a2d36f7b/cantools/database/can/formats/dbc.py#L1191

As can be seen from:
https://github.com/cantools/cantools/blob/92f4c020df296402040eb84139bf7678a2d36f7b/cantools/database/can/formats/dbc.py#L309-L311

Can easily be reproduced by adding a sample line, with the above format, to:
https://github.com/cantools/cantools/blob/92f4c020df296402040eb84139bf7678a2d36f7b/tests/files/dbc/attributes_relation.dbc#L61

Issue was introduced with #430.

juleq, 2022-08-19:
@malneni Could you please have a look at this? I do not see a discrimination between rel attributes associated with messages vs. with signals in _load_attributes_rel. If this is because it is not implemented, it still should not break in this case. If you do not plan to implement, this still should be fixed and a test case added that ensures this break is gone. 

Thanks.

@CodenameReese I think @malneni might have implemented just what he needed in his use case. This break was probably not anticipated. If you want to take a shot at it, be our guest :). Does it break for you without even requiring the rel attribute or do you use rel attributes?

malneni, 2022-08-21:
@juleq, I could not create any rel attributes for Messages with tools that I have. I do not know dbc format for that.
I will try to fix the break

CodenameReese, 2022-08-22:
I could also create a PR, but I think @malneni is already on it 😉

I threw together a quick sample dbc, I hope this helps with testing:

```
VERSION ""

NS_ : 
	NS_DESC_
	CM_
	BA_DEF_
	BA_
	VAL_
	CAT_DEF_
	CAT_
	FILTER
	BA_DEF_DEF_
	EV_DATA_
	ENVVAR_DATA_
	SGTYPE_
	SGTYPE_VAL_
	BA_DEF_SGTYPE_
	BA_SGTYPE_
	SIG_TYPE_REF_
	VAL_TABLE_
	SIG_GROUP_
	SIG_VALTYPE_
	SIGTYPE_VALTYPE_
	BO_TX_BU_
	BA_DEF_REL_
	BA_REL_
	BA_DEF_DEF_REL_
	BU_SG_REL_
	BU_EV_REL_
	BU_BO_REL_
	SG_MUL_VAL_

BS_:

BU_: ECU1 ECU2

BO_ 1 message_1: 7 ECU1
 SG_ signal_1 : 0|8@1+ (1,0) [0|100] "" ECU2

BA_DEF_REL_ BU_BO_REL_  "MsgProject" ENUM  "A","B","C";
BA_DEF_DEF_REL_ "MsgProject" "A";
BA_REL_ "MsgProject" BU_BO_REL_ ECU1 1 2;
VAL_ 1 signal_1 255 "Invalid" ;

```

malneni, 2022-09-01:
@CodenameReese, Updated code is now merged to fix this issue

CodenameReese, 2022-09-05:
@malneni Thank you very much for the quick fix!