cantools/cantools #478: can't load dbc with utf-8 characters

fr0zenrain, 2022-09-01:
db = cantools.database.load_file(sys.argv[1]) throw erros
cantools.database.UnsupportedDatabaseFormatError: DBC: "Invalid syntax at line 28, column 10: " SG_ TBOX>>!<<´ÎÈí¼þ°æ±¾ºÅ$_W : 15|8@0+(1,0)[0|255] "" Vector__XXX""

it seems dbc.py's  token_specs regex just only identify ascii characters, so it has a plan to fix utf-8 issue? thanks!

juleq, 2022-09-01:
Could you provide a sample file? Does Vector CANdb++ open that file?

andlaus, 2022-09-01:
you could also try if
```python
db = cantools.database.load_file(sys.argv[1], encoding="utf8")
```
does the trick...

fr0zenrain, 2022-09-01:
> you could also try if
> 
> ```python
> db = cantools.database.load_file(sys.argv[1], encoding="utf8")
> ```
> 
> does the trick...

not work

fr0zenrain, 2022-09-01:
with an example dbc file, and find the "SG_" line, replace with other language ,i'm not sure wether dbc specification is support or not. i found some software support it,but CANdb++ 3.1 not support.
SG_ TBOXVersion : 15|8@0+(1,0)[0|255] "" Vector__XXX
SG_ TBOX版本 : 15|8@0+(1,0)[0|255] "" Vector__XXX

juleq, 2023-01-04:
Since CANdb++ is authoritative to the dbc file format, I think this will have to be considered unsupported.

closed 2023-01-04 by juleq