cantools/cantools #480: Why CAN DBC parsed object node object don't contain message object or message don't contain node attribute

niuzijia, 2022-09-07:
Why CAN DBC parsed object node object don't contain message object or message don't contain node attribute. It seem that Nodes should cotain multiple message object , just like CANoe interactive layer

juleq, 2022-09-07:
That's probably because you were not here to implement that before 😉.

I would agree that having a messages collection in the node object would be a convenient way to implement residual bus simulations as done with Vectors interactive layer.

Concerning general (inconvenient) availability of this information: Have you checked the "senders" attribute of the message object?

So you can help yourself by creating and attaching a messages collection to each node after loading your dbc.

If you want to contribute this feature, you could add this step as some post dbc load activity. Please check the style guide and provide tests that ensure we keep this feature working properly.

niuzijia, 2022-09-07:
Thank you for your patience, I thought it was caused by some other reason. I will try to add this feature later😉