cantools/cantools #483: Read issue on dbc EV_

cartooh, 2022-09-19:
when reading .dbc file with EV_, if multiple access_nodes are defined, they cannot be read.

The definition of EV_ is as follows

> environment_variable = 'EV_' env_var_name ':' env_var_type '[' minimum '|' maximum ']' unit initial_value ev_id access_type access_node {',' access_node } ';' ; 

http://mcu.so/Microcontroller/Automotive/DBC_File_Format_Documentation.pdf

https://github.com/cantools/cantools/blob/master/cantools/database/can/formats/dbc.py#L263
```
        environment_variable = Sequence(
            'EV_', 'WORD', ':', 'NUMBER',
            '[', 'NUMBER', '|', 'NUMBER', ']',
            'STRING', 'NUMBER', 'NUMBER', 'WORD', 'WORD', ';')
```

It should be as follows.
```
        environment_variable = Sequence(
            'EV_', 'WORD', ':', 'NUMBER',
            '[', 'NUMBER', '|', 'NUMBER', ']',
            'STRING', 'NUMBER', 'NUMBER', 'WORD', 'WORD', ZeroOrMore(Sequence(',', 'WORD')), ';')
```

juleq, 2022-09-19:
Hi,

PRs are always welcome. Adhering to the recommended code style would be required.

To add support for that, the way would be:
- Add it to the grammar
- Add it to the load function
- Find a place to store it
- Optional: Make it available through some API
- Add it to the dump function
- Add tests to ensure it continues to be maintained 

The reason for it not being in is probably a low number of people requiring it. And I think Vector has  retired EVs in recent versions. But you should fact-check that :).

cartooh, 2022-09-22:
Thanks for the reply.

Indeed, it was mentioned in the CANdb++ help that EVs will not be supported in the future.

I don't use EV definitions, I just didn't want to modify thousands of lines of old DBC files to avoid syntax errors.

If Vector does not support it, I will withdraw this issue.

Other proprietary extensions are needed and I have already customized mine.

juleq, 2022-09-22:
I have labeled this as a possible enhancement. I think it is a reasonable request to be compatible with older versions of the dbc format (I assume cantools errors out due to grammar mismatch at the moment?). 

I cannot add this myself at the moment, no time to spare.

One approach to support is to do it in full (so also adding it to the data model), another one would be to just recognize it as content that will not be processed further and save it to be able to support loading and dumping transparently at least.

closed 2026-09-15 by andlaus