cantools/cantools #491: how to load arxml

Lit-fish-auto, 2022-10-11:
Generally, the arxml format of CANFD is   Frame :: Container-PDU- pdu :: I-signal-PDU :: Signal
If arxml is in the above format, thenThe method of  load_ file（abc.arxml） cannot read the signal.

Above, how should this form of arxml be read? or  is the bug ？

Lit-fish-auto, 2022-10-13:
could you plaese send  system-4.2.arxml  to zg6543466@126.com?
Sorry, I don't have permission to download this file  system-4.2.arxml . Can you send the document to my email  zg6543466@126.com ? I will be very grateful.
I want to open the arxml in the same load mode and Vcetor ，for comparison to check whether the same is correct.
@andlaus

Lit-fish-auto, 2022-10-13:
[image]
[image]
[image]
@andlaus

Lit-fish-auto, 2022-10-13:
Open systerm-4.2 with vector, and an error occurs.
The version of my local 123abc.arxml file is 4.2
After opening 123abc.arxml  with vector, you will find that FrP04 contains Container-I-PDU, and also contains signal-I-PDU
However, through cantools, the signal contained in the signal-I-PDU cannot be read。

andlaus, 2022-10-13:
yeah, `system-4.2.arxml` triggers a PDU size warning with Vector's AUTOSAR explorer. That said, I cannot help you with the other file if I don't have access to it.

but as said above: container messages do not exhibit any signals themselves (only their child messages do), so it is not surprising that the snippet you show above does not work.

ChaselP, 2022-10-21:
@Lit-fish-auto 
Hi Bro, try that:
`message.is_container` and `message.contained_messages`
Maybe we are colleagues of SAIC, if so we can communicate offline.

Lit-fish-auto, 2022-11-28:
对的，可能我们是同事。
你是EE的吗?

---- Replied Message ----
| From | ***@***.***> |
| Date | 10/21/2022 10:09 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [cantools/cantools] how to load arxml (Issue #491) |

@Lit-fish-auto
Hi Bro, try that:
message.is_container and message.contained_messages
Maybe we are colleagues of SAIC, if so we can communicate offline.

luojun6, 2024-08-21:
> 对的，可能我们是同事。 你是EE的吗?
> […](#)

Interesting！