cantools/cantools #497: Improperly decoded byte array

MichaelBMiner, 2022-10-25:
Hello. I am using DBCPPP parser to encode a string and send it to a python script running on a Pi to receive and decode my CAN messages. I am using the J1939 standard and I have a DBC entry for my string which is as follows:

```
BO_ 2566059006 PGN_65523_SMG_EVCC_ID: 8 EVCC
 SG_ EVCC_ID : 0|64@1+ (1,0) [0|255] "" EV
```

I am sending the string "smg12345" 8 bytes a full payload. When I convert this to a uint64_t I get the value 7435779655266009088.

When I RX the data I get the following:
```
Message Timestamp: 1666717113.815678    ID: 18f2f3fe    X                DLC:  8    00 00 35 34 33 32 31 67     Channel: can0
Message Data bytearray(b'\x00\x0054321g')
Data {'EVCC_ID': 7435779655266009088}
```

You can see that the data is the same, but the byte array is missing 2 values, s and m. I have verified that my DBC files on the sending and receiving devices are the same. I see this issue no matter how many bytes I send there is always data missing in the byte array. Can you please take a look and see if you can provide any suggestions.

andlaus, 2022-10-26:
can you share a DBC file where this happens? (it is rather uncommon to send strings in messages, so this is probably untested functionality...)

janaranjanahl, 2022-10-26:
this is the dbc entry

```
BO_ 2566059006 PGN_65523_SMG_EVCC_ID: 8 EVCC
 SG_ EVCC_ID : 0|64@1+ (1,0) [0|255] "" EV
```

zariiii9003, 2022-10-27:
> I am sending the string "smg12345" 8 bytes a full payload. When I convert this to a uint64_t I get the value 7435779655266009088.

Are you sure?
```python3
>>> int.from_bytes(b"smg12345", byteorder="big")
8317417547816449077
>>> int.from_bytes(b"smg12345", byteorder="little")
3833745473469377907
```

MichaelBMiner, 2022-11-01:
You are correct, my input may be incorrect for this example. My issue still remains. I sent smg12345 but only received 54321g. Two bytes were not decoded and displayed. Even though the raw data comes across correct!

zariiii9003, 2022-11-01:
There is no smg12345 in the raw data of your CAN message:

Message Timestamp: 1666717113.815678    ID: 18f2f3fe    X                DLC:  8    `00 00 35 34 33 32 31 67`     Channel: can0

```python3
int.from_bytes(bytes.fromhex("00 00 35 34 33 32 31 67"), byteorder="little")
7435779655266009088
```

I do not see any decoding errors here.

closed 2023-01-04 by juleq