cantools/cantools #499: The signal.comment can

songzuolong, 2022-10-26:
Python parses dbc files, for Chinese garbled characters in siganl.comment

andlaus, 2022-10-26:
if your file is does not use UTF-8 encoding, you need to explicitly pass it via the `encoding` parameter for [`load_file()`](https://github.com/cantools/cantools/blob/master/cantools/database/__init__.py#L104)

songzuolong, 2022-10-27:
my dbc file: VAL_ 345 WiprHoodOpenInde 0 "关闭提醒" 1 "触发提醒" ;
use the db = cantools.database.load_file("c:\\Users\\szl\\Desktop\\DBC\\MS11_DBC.dbc",encoding="utf-8") 
but still garbled。
ǰ��δ��Ԥ����Ť����Ч��

songzuolong, 2022-10-27:
I have sloved this problem， 
you are right， i need to add the encoding parameter for load_file().
I.e.  db = cantools.database.load_file("c:\Users\szl\Desktop\DBC\MS11_DBC.dbc",encoding="gbk")

songzuolong, 2022-10-27:
> if your file is does not use UTF-8 encoding, you need to explicitly pass it via the `encoding` parameter for [`load_file()`](https://github.com/cantools/cantools/blob/master/cantools/database/__init__.py#L104)

thanks

closed 2022-10-27 by andlaus