cantools/cantools #503: Unable to decode j1939 frames

hbertran, 2022-11-23:
Hello, 

I am using cantools to receive and decode j1939 can frames that are replayed using BUSMASTER, but when calling the method decode_message, I get a KeyError.
The arbitration id that I receive in my python code does not match the one in my DBC file, but BUSMASTER manages to decode messages when emitting them (using that same DBC), so I suppose I'm missing something to do with the arbitration id in order to find my message in the DBC file, but I have no clue what it is.

Does anyone have an idea to solve this problem ?

juleq, 2022-11-23:
You could post the offending id in hex and as found in the dbc file (decimal, open with text editor). Sometimes the extended bit is missing.

Another point you could check is if destination/source addresses have changed. J1939 uses the least significant id bytes for them.

hbertran, 2022-11-23:
Here's an example : 
I receive the arbitration ID 0xc0000e6, which is 201326822 in decimal and matches the one that appears in my BUSMASTER window.
In my DBC file, the ID of this message is 2348875518.
I also opened the DBC file with KVaser's Database Editor and the ID that it displays for my message is 201391870 (or c00fefe in hexadecimal)
In addition, the source and destination given by BUSMASTER are E6 and 00

juleq, 2022-11-23:
201326822 Is not an extended id in dbc format. Add  1 << 31 to it and it will be extended. So 2147483648 + 201326822.

juleq, 2022-11-23:
I do not know which tool was used to write the dbc, but either the message has not been configured to ext id or the tool encodes extended ids incorrectly.

hbertran, 2022-11-23:
I tried adding 2147483648 to the received arbitration Ids but I'm still getting key errors. I don't know either which tool was used to write the dbc file, but I don't think the problem comes from this file since BUSMASTER manages to decode the messages with this same DBC

juleq, 2022-11-23:
Please double-check. 2348810470 should be the id found in the dbc. Pretty sure that is your issue. dbc won’t have ext ids any other way.

Busmaster could infer that the id is to big to be a std id and correct it.

hbertran, 2022-11-23:
Checked again, same result. When experimenting I noticed that in order for cantools to find my message in the dbc I must use the id that is displayed in KVaser Database Editor (which is 201391870). I don't know how to get this value from the 201326822  that I receive.

juleq, 2022-11-24:
First id you mentioned:
0x0C00FEFE —(set ext id bit)—> 0x8C00FEFE —(hex to dec for dbc file usage)—>  [2348875518](tel:2348875518)

Second id you mentioned: 0x0C0000E6 —(set ext id bit)—> 0x8C0000E6 —(hex to dec for dbc file usage)—> [2348810470](tel:2348810470)
 
Please note that source and destination addresses between the two ids differ:

Source addresses (byte 0 of id):
                First 0xFE, second 0xE6

Destination addresses (byte 1 of id in case these PGNs use that field as destination, not always the case):
                First 0xFE, second 0x00

juleq, 2022-11-24:
Busmaster has support for J1939 semantics. Cantools does not. You need to change the ids. You can either change them to the incoming id values in the dbc, or change the message id with Python after loading it. You probably have to call the refresh function on the db because the ids are used as keys in dicts and need to be updated after changing.

hbertran, 2022-11-24:
How can I change the ids with Python ? And to what value should I set them ?

juleq, 2022-11-24:
If you want to do it in python, you can check the docs, specifically the message and the database objects.

You would set them to the ids that you see on the bus. You can use a hex literal to set it (without ext bit). Also check that the message is set to extended.

Then call refresh on the database.

There is some j1939 support, but I to not know how far that goes. If your dbc attributes are set correctly, there might be some functions to unpack the id elements. But I am not sure that masked reception (which would be required to automatically take in frames of the same type from e.g. different source addresses) is implemented for j1939. Never used it.

closed 2023-01-04 by juleq