cantools/cantools #504: Unable decode the dbc file

smpshehan, 2022-11-25:
I got this dbc file from csselectronics. I am using a simple python code to decode a message. But I get error saying that "KeyError: 4512038398"
 File "C:\DBC\test.py", line 17, in <module>
    db.decode_message(4512038398,b'FFFFFF80B2FFFFFF')
  File "C:\Local\Programs\Python\Python311\Lib\site-packages\cantools\database\can\database.py", line 485, in decode_message
    message = self._frame_id_to_message[frame_id_or_name]
              ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
KeyError: 4512038398

I know if it is an extended id, then I have to add 0x80000000 to the frame id. I tried that too. Still that frame id cannot be found in the dbc file to decode my message. Is the following code correct?

```
db = cantools.database.load_file('./css2.dbc')
db.decode_message(4512038398,b'FFFFFF80B2FFFFFF')
```
How do I get this dbc file working from csselectronics?

juleq, 2022-11-25:
Dec 4512038398 is definitely wrong. That would be 0x10CF03DFE. Looks like someone tried to set the most significant bit and went left one too far.

smpshehan, 2022-11-25:
Thank you. My question is, what is the format to pass this information to
decode? Hex or decimal? Is my above Python code correct?

juleq, 2022-11-25:
Python uses no prefix for decimal literals and 0x as the prefix for hex literals, so you can do both.

Since your id is wrong, you get a key error because the id is used as key in the message dict/map.

smpshehan, 2022-11-26:
what about the message part?
db.decode_message(4512038398,b'FFFFFF80B2FFFFFF')
Do I need to use b' as prefix or /xFF/xFF etc? Is my above code correct?

juleq, 2022-11-26:
Looks incorrect. You can read up on byte literals here:
https://peps.python.org/pep-3112/

juleq, 2022-11-26:
You can always open a Python REPL and play around with it:
`
>>> yours = b'FFFFFF80B2FFFFFF'
>>> oneCorrectVariant = bytes([0xff,0xff,0xff,0x80,0xb2,0xff,0xff,0xff])

>>> list(yours)
[70, 70, 70, 70, 70, 70, 56, 48, 66, 50, 70, 70, 70,
 70, 70, 70]
>>> list(oneCorrectVariant)
[255, 255, 255, 128, 178, 255, 255, 255]

`

andlaus, 2022-11-27:
>  Is my above Python code correct?

depends on what you want to do: you're trying to decode the 16-byte ASCII string "FFFFFF80B2FFFFFF", instead of the 8-byte binary string which you probably want. What you probably want to do is something like this:
```python
db.decode_message(4512038398, bytes.fromhex('FFFFFF80B2FFFFFF'))
```

smpshehan, 2022-11-28:
Cool, thanks for all the answers. I know sometimes the CAN id (key id) is an extended id. So I have to add some bits. Otherwise I will not be able to find the key in dbc file. What is the rule for that? How do I know when I have to add some bits to the key id?

smpshehan, 2022-12-01:
I have this information in my dbc file.

`BO_ 2365484286 TCO1: 8 Vector__XXX
 SG_ DirectionIndicator : 30|2@1+ (1,0) [0|3] "" Vector__XXX
 SG_ Driver1TimeRelatedStates : 8|4@1+ (1,0) [0|15] "" Vector__XXX
 SG_ Driver1WorkingState : 0|3@1+ (1,0) [0|7] "" Vector__XXX
 SG_ Driver2TimeRelatedStates : 16|4@1+ (1,0) [0|15] "" Vector__XXX
 SG_ Driver2WorkingState : 3|3@1+ (1,0) [0|7] "" Vector__XXX
 SG_ DriverCardDriver1 : 12|2@1+ (1,0) [0|3] "" Vector__XXX
 SG_ DriverCardDriver2 : 20|2@1+ (1,0) [0|3] "" Vector__XXX
 SG_ HandlingInformation : 26|2@1+ (1,0) [0|3] "" Vector__XXX
 SG_ SystemEvent : 24|2@1+ (1,0) [0|3] "" Vector__XXX
 SG_ TachographOutputShaftSpeed : 32|16@1+ (0.125,0) [0|8031.875] "rpm" Vector__XXX
 SG_ TachographPerformance : 28|2@1+ (1,0) [0|3] "" Vector__XXX
 SG_ TachographVehicleSpeed : 48|16@1+ (0.00390625,0) [0|250.99609375] "km/h" Vector__XXX
 SG_ VehicleMotion : 6|2@1+ (1,0) [0|3] "" Vector__XXX
 SG_ VehicleOverspeed : 14|2@1+ (1,0) [0|3] "" Vector__XXX`

When I use the command 
`candb.decode_message(2365484286, bytes.fromhex(FFFFFFFFFFFF00D1))`
it can't find the id. I get error  KeyError: '2365484286'
But if I use the frame name as follows, it can decode
`candb.decode_message(2365484286, bytes.fromhex(FFFFFFFFFFFF00D1))`
Can someone please explain what am I doing wrong?

juleq, 2022-12-01:
You need to remove the ext bit from the dec value. dbc encodes it that way. But cantools has a separat boolean flag to indicate that (as it should be).

Just use 0x0CFE6CFE.

smpshehan, 2022-12-01:
These are sample values I want to decode. First part is the CAN Id and next part is the actual data.
CF00400#FFFFFFC080FFFFFF
CFE6C00#FFFFFFFFFFFF00D1
CF00300#FF12FFFFFFFFFFFF
18FEF600#FFD3FFFFFFFFFFFF
18FEEE00#9350FFFFFFFFFFFF
CFE6C00#FFFFFFFFFFFF00D1

How do I convert the CAN Id(frame id) part? e.g: CFE6C00

juleq, 2022-12-01:
2365484286 = 0x8CFE6CFE

And then you remove the most significant bit, e.g by subtracting 0x80000000.

smpshehan, 2022-12-01:
ok, what is the rule for this? Do I have to do this for every frame id?

juleq, 2022-12-01:
And if you want to get the decimal value as in the dbc file, you would add 0x80000000 and do hex to dec.

juleq, 2022-12-01:
Rule is just add the hex literal prefix 0x and you are good.

smpshehan, 2022-12-01:
then I get KeyError: 218000384 for 0x0CFE6C00

juleq, 2022-12-01:
How did you load the dbc? Can you post the message definition of this message? The BO_?

smpshehan, 2022-12-01:
yes J1939. I am still confused. How do I decode CFE6C00#FFFFFFFFFFFF00D1 ?

candb.decode_message(0x0CFE6C00, bytes.fromhex('FFFFFFFFFFFF00D1')) gives me KeyError message

smpshehan, 2022-12-01:
Is the decoding different for J1939? We got this dbc file from csselectronics

juleq, 2022-12-01:
Depends on what you want to do. If you just want to decode a few frames, you can just adapt the ids in the dbc and use cantools. If you need more elaborate j1939 support, another tool might be more useful to you.

smpshehan, 2022-12-02:
Thanks for all your help @juleq. So I had replace the last 2 bytes with FE. So CFE6C00 => CFE6CFE
That will give me the matching frame id in my dbc file, which is a J1939 CAN dbc file

closed 2023-01-04 by juleq

acode-x, 2023-01-07:
I have the similar problem, but it may not be related to `cantools` I assume.

Here is a sample data I used from csselectronics,
```log
TimestampEpoch;BusChannel;ID;IDE;DLC;DataLength;Dir;EDL;BRS;DataBytes
1578926829.807850;1;18FEDB01;1;8;8;0;0;0;A354A354A354FFFF
```
```bash
cansend vcan0 18FEDB01#A354A354A354FFFF
```

`18FEDB01` refers to `Engine Fluid Level/Pressure 2` as per https://www.csselectronics.com/pages/j1939-pgn-conversion-tool

If I see the equivalent of `Engine Fluid Level/Pressure 2` in J1939 DBC from https://github.com/nberlette/canbus/blob/main/dbc/j1939.dbc it says `18FEDFE`

Since these are not matching, cantools decode_message() would fail.
But am not too sure which one is wrong.

juleq, 2023-01-07:
18FEDFE would be surprising, because the B is missing (compared to your other example).

Otherwise: One of your examples uses a J1939 source address of 0x01 and the other uses 0xFE. To decode without J1939 support, you just need to make them match. Either by changing your input data or your dbc.

acode-x, 2023-01-07:
I stand corrected.
I was little confused with the different terminologies.
I was able to fix the same by following changes.

Firstly, I assume frame_id is same as PGN
Secondly, I understand CAN ID is different from DBC ID but both result in same PGN

[image]

Thirdly, I changed the message parsing logic in cantools/database/can/formats/dbc.py
```python
# frame_id = frame_id_dbc & 0x7fffffff # <--- old logic
priority_pgn = frame_id_dbc >> 8  # remove 8bits source address
frame_id = pgn = (
    priority_pgn & 0b000000111111111111111111
)  # In J1939, DBC ID is 32bit? remove 3bits(32-29CAN_ID) + 3bits priority
```

And here is the test code,
```python
import can
import cantools

def get_pgn_from_can_id(can_id: int) -> int:
    priority_pgn = can_id >> 8
    return priority_pgn & 0b000111111111111111111  # In J1939, CAN ID is 29bit

mydb = cantools.db.load_file("./j1939-large.dbc")

print("Listening on vcan0!")
can_bus = can.interface.Bus("vcan0", bustype="socketcan")
message = can_bus.recv()

decoded_message = mydb.decode_message(
    frame_id_or_name=get_pgn_from_can_id(message.arbitration_id),
    data=message.data,
    decode_choices=False,
)
print(decoded_message)
```

```bash
> cansend vcan0 18FEDB01#A354A354A354FFFF
```

Above works as expected.