cantools/cantools #506: [DBC Question] Does dbc format support "2 parents" Extended Multiplexing?

JiapengLi, 2022-11-28:
Sorry, this is a not an issue related to cantools.It is a question about dbc file format. 

It is about how to use DBC `SG_MUL_VAL_` key word to define Extended Multiplexing.

I am hoping a dbc description file like below to work. Which supports an Extended Multiplexing by an AND logic Multiplexing of SERVICE and PID. But it doesn't, I test with cantools, it seems cantools doesn't support it.

```dbc
BO_ 2024 RX: 64 ECU
 SG_ RXLEN : 0|8@1+ (1,0) [0|0] "" TOOL
 SG_ SERVICE M : 8|4@1+ (1,0) [0|0] "" TOOL
 SG_ RXACK : 12|4@1+ (1,0) [0|0] "" TOOL
 SG_ PID M : 16|8@1+ (1,0) [0|0] "" TOOL
 SG_ S01_PID00_SupportedIDs_01_20 m1m0 : 24|32@1+ (1,0) [0|0] "" TOOL
 SG_ S02_PID00_SupportedIDs_01_20 m2m0 : 24|32@1+ (1,0) [0|0] "" TOOL

SG_MUL_VAL_ 2024 S01_PID00_SupportedIDs_01_20 SERVICE 1-1;
SG_MUL_VAL_ 2024 S01_PID00_SupportedIDs_01_20 PID 0-0;
SG_MUL_VAL_ 2024 S02_PID00_SupportedIDs_01_20 SERVICE 2-2;
SG_MUL_VAL_ 2024 S02_PID00_SupportedIDs_01_20 PID 0-0;
```

My current workaround is like below, which renames PID to different names to force creating a multiplexing route, this makes the file very tedious and break things, it leads PID filed to be parsed as a different name, in logic the PID field should be parsed as the same name with different value.

```
BO_ 2024 RX: 64 ECU
 SG_ RXLEN : 0|8@1+ (1,0) [0|0] "" TOOL
 SG_ SERVICE M : 8|4@1+ (1,0) [0|0] "" TOOL
 SG_ RXACK : 12|4@1+ (1,0) [0|0] "" TOOL
 SG_ S01_PID m1M : 16|8@1+ (1,0) [0|0] "" TOOL
 SG_ S02_PID m2M : 16|8@1+ (1,0) [0|0] "" TOOL
 SG_ S03_PID m3M : 16|8@1+ (1,0) [0|0] "" TOOL
 SG_ S04_PID m4M : 16|8@1+ (1,0) [0|0] "" TOOL
 SG_ S05_PID m5M : 16|8@1+ (1,0) [0|0] "" TOOL
 SG_ S06_PID m6M : 16|8@1+ (1,0) [0|0] "" TOOL
 SG_ S07_PID m7M : 16|8@1+ (1,0) [0|0] "" TOOL
 SG_ S08_PID m8M : 16|8@1+ (1,0) [0|0] "" TOOL
 SG_ S09_PID m9M : 16|8@1+ (1,0) [0|0] "" TOOL
 SG_ S0A_PID m10M : 16|8@1+ (1,0) [0|0] "" TOOL
 SG_ S01_PID00_SupportedIDs_01_20 m0 : 24|32@1+ (1,0) [0|0] "" TOOL
 SG_ S02_PID00_SupportedIDs_01_20 m0 : 24|32@1+ (1,0) [0|0] "" TOOL

SG_MUL_VAL_ 2024 S01_PID SERVICE 1-1;
SG_MUL_VAL_ 2024 S02_PID SERVICE 2-2;
SG_MUL_VAL_ 2024 S03_PID SERVICE 3-3;
SG_MUL_VAL_ 2024 S04_PID SERVICE 4-4;
SG_MUL_VAL_ 2024 S05_PID SERVICE 5-5;
SG_MUL_VAL_ 2024 S06_PID SERVICE 6-6;
SG_MUL_VAL_ 2024 S07_PID SERVICE 7-7;
SG_MUL_VAL_ 2024 S08_PID SERVICE 8-8;
SG_MUL_VAL_ 2024 S09_PID SERVICE 9-9;
SG_MUL_VAL_ 2024 S0A_PID SERVICE 10-10;
SG_MUL_VAL_ 2024 S01_PID00_SupportedIDs_01_20 S01_PID 0-0;
SG_MUL_VAL_ 2024 S02_PID00_SupportedIDs_01_20 S02_PID 0-0;
SG_MUL_VAL_ 2024 S05_PID80_SupportedIDs_81_A0 S05_PID 128-128;
SG_MUL_VAL_ 2024 S06_PID80_SupportedIDs_81_A0 S06_PID 128-128;
SG_MUL_VAL_ 2024 S07_PID80_SupportedIDs_81_A0 S07_PID 128-128;
SG_MUL_VAL_ 2024 S08_PID80_SupportedIDs_81_A0 S08_PID 128-128;
SG_MUL_VAL_ 2024 S09_PID80_SupportedIDs_81_A0 S09_PID 128-128;
SG_MUL_VAL_ 2024 S0A_PID80_SupportedIDs_81_A0 S0A_PID 128-128;
SG_MUL_VAL_ 2024 S01_PIDA0_SupportedIDs_A1_C0 S01_PID 160-160;
SG_MUL_VAL_ 2024 S02_PIDA0_SupportedIDs_A1_C0 S02_PID 160-160;
SG_MUL_VAL_ 2024 S03_PIDA0_SupportedIDs_A1_C0 S03_PID 160-160;
SG_MUL_VAL_ 2024 S04_PIDA0_SupportedIDs_A1_C0 S04_PID 160-160;
SG_MUL_VAL_ 2024 S05_PIDA0_SupportedIDs_A1_C0 S05_PID 160-160;
SG_MUL_VAL_ 2024 S06_PIDA0_SupportedIDs_A1_C0 S06_PID 160-160;
SG_MUL_VAL_ 2024 S07_PIDA0_SupportedIDs_A1_C0 S07_PID 160-160;
SG_MUL_VAL_ 2024 S08_PIDA0_SupportedIDs_A1_C0 S08_PID 160-160;
SG_MUL_VAL_ 2024 S09_PIDA0_SupportedIDs_A1_C0 S09_PID 160-160;
SG_MUL_VAL_ 2024 S0A_PIDA0_SupportedIDs_A1_C0 S0A_PID 160-160;
SG_MUL_VAL_ 2024 S01_PIDC0_SupportedIDs_C1_E0 S01_PID 192-192;
SG_MUL_VAL_ 2024 S02_PIDC0_SupportedIDs_C1_E0 S02_PID 192-192;
SG_MUL_VAL_ 2024 S03_PIDC0_SupportedIDs_C1_E0 S03_PID 192-192;
SG_MUL_VAL_ 2024 S04_PIDC0_SupportedIDs_C1_E0 S04_PID 192-192;
SG_MUL_VAL_ 2024 S05_PIDC0_SupportedIDs_C1_E0 S05_PID 192-192;
SG_MUL_VAL_ 2024 S06_PIDC0_SupportedIDs_C1_E0 S06_PID 192-192;
SG_MUL_VAL_ 2024 S07_PIDC0_SupportedIDs_C1_E0 S07_PID 192-192;
SG_MUL_VAL_ 2024 S08_PIDC0_SupportedIDs_C1_E0 S08_PID 192-192;
SG_MUL_VAL_ 2024 S09_PIDC0_SupportedIDs_C1_E0 S09_PID 192-192;
SG_MUL_VAL_ 2024 S0A_PIDC0_SupportedIDs_C1_E0 S0A_PID 192-192;
```

PS: This question is come from a dbc project what I am doing. Creating a dbc file to fully support obdii (iso15031-5 and J1979DA). Repo is here https://github.com/jiapengLi/dbc

andlaus, 2022-11-28:
I guess you mean nested multiplexers? Whilst they are supported by both the DBC file format and cantools, they are -- as you've already noticed -- a pretty messy affair and the code in cantools is not very battle-hardened.

If you want to create a description of OBD-II for usage over ISO-TP, I would not recommend trying to do this using DBC because it is a file format specialized for regular signals-based communication. Instead, I'd like to recommend to use a file format adapted to the needs diagnostic descriptions. (I might be biased, though: I'm also involved in [odxtools](https://github.com/mercedes-benz/odxtools), an analog to cantools for diagnostics using the [ODX / ISO22901-1](https://www.embitel.com/blog/embedded-blog/what-is-open-diagnostics-exchange-odx ) specifications.)

JiapengLi, 2022-11-29:
Hi @andlaus, I think what I need is a 2 same level multiplexors ( service and pid). I want to keep the name PID for different services. To do so, application layer just need check signals named SERVICE and PID. Ease the sw design

andlaus, 2022-11-29:
> I think what I need is a 2 same level multiplexors ( service and pid). 

in the context of cantools, that is a nested multiplexer. UDS and OBD-II telegrams look something like this:

```
+-------+-------+--------+--------+--------+----------+
| sid   | pid   | param1 | param2 | ...    |  param_n |
+-------+-------+--------+--------+--------+----------+
```

where the parameters depend on th PID and SID. while it is not completely impossible to specify this using nested multiplexers, such structures are *very* inconvenient to specify using DBC (or any other format supported by cantools). you have been warned ;)

JiapengLi, 2022-11-30:
@andlaus Thank you for your prompt reply. Agree with you, `sid` and `pid` are actually at the same level, but dbc (not sure it is specification related or implemetation related) seems not support it. 

My current solution looks like below, have to rename or create a new pid variable, because same name means same param structure. 
```
 +-------+-----------+--------+--------+--------+----------+
 | sid   |  s01_pid  | param1 | param2 | ...    |  param_n |
 +-------+-----------+--------+--------+--------+----------+
 +-------+-----------+--------+--------+--------+----------+
 | sid   |  s02_pid  | param1 | param2 | ...    |  param_n |
 +-------+-----------+--------+--------+--------+----------+
```
Another option maybe to merge pid and sid, but it is also inconvenient. ;(

BTW, I take a look at the odxtools roughly, is very interesting. Good job! I may go deep into it after I finish odbii dbc library.