cantools/cantools #510: AttributeError: module 'can' has no attribute 'Listener'

shotrecommendation, 2022-12-12:
Hello,

I am quite a begginner in coding so I am not entirely sure if the issue is caused on my end or is it related to the project. Nevertheless:

1. On import:
`from cantools import database`
I have received an error:
`Traceback (most recent call last):
  File "D:\Users\wolujewicz_w\H2analyzer\analyser.py", line 2, in <module>
    from conversion.dbc import create_dbc_database
  File "D:\Users\wolujewicz_w\H2analyzer\conversion\dbc.py", line 1, in <module>
    from cantools import database
  File "D:\Users\wolujewicz_w\H2analyzer\venv\lib\site-packages\cantools\__init__.py", line 7, in <module>
    from . import tester
  File "D:\Users\wolujewicz_w\H2analyzer\venv\lib\site-packages\cantools\tester.py", line 28, in <module>
    class Listener(can.Listener):
AttributeError: module 'can' has no attribute 'Listener'`
2. I could not find a similar issue, so in my venv I have edited the tester.py file so that it imports the Listener directly:
`from can.listener import Listener`
and edited the 28th line of the tester.py file accordingly to this:
`class Listener(Listener): (...)`

Obviously it is very ugly solution and I would like to fix it properly. Can I get an advice on that? Should it be fixed in cantools itself or is it rather an issue with my own setup?

juleq, 2022-12-12:
You might have installed a Python package called „can“ by accident. That clashes with the name of a sub-module.

Use pip to remove the can package (some color tool…).

shotrecommendation, 2022-12-12:
Unfortunately that is not the case. I do have a can package installed in my venv but it is a can package that comes with python itself and it actually is used for can bus stuff.

juleq, 2022-12-12:
What does „pip show can“ return?

shotrecommendation, 2022-12-12:
Oh man, you were right. Here comes my inexpierience I guess. There has been tis "color tool" you have mentioned. 

Nevertheless, I have used "pip uninstall can", now the command "pip show can" returns:
>WARNING: Package(s) not found: can

I undoed my change in the tester.py file as well. The issue still exists (the initial traceback that I have shared)

shotrecommendation, 2022-12-12:
I have solved the issue thanks to @juleq. I created a new venv and installed all necessary packages once again, this time without the problematic and unnecessary "can" package. After that, the abovementioned error has disappeared.