cantools/cantools #517: Multiplexed signals with cantools.tester

markrages, 2023-01-24:
I am having trouble getting the tester to work with multiplexed signals.  Here is an example:

```python
#!/usr/bin/env python3

dbc_file = 'cantools/tests/files/dbc/multiplex_2.dbc'

import can, cantools

with can.Bus(channel = "vcan0", bustype = "socketcan") as bus:

    db = cantools.database.load_file(dbc_file)

    # dirty hack to fake a dut name
    if not db.nodes:
        class FakeNode: name='fake'

        db.nodes.append(FakeNode())

    node_names = [node.name for node in db.nodes]
    dut = node_names[0]

    bus_names = [bus.name for bus in db.buses]
    bus_name = bus_names[0] if bus_names else None

    tester = cantools.tester.Tester(dut, db, bus, bus_name=bus_name)
```
which results in this traceback:
```
Traceback (most recent call last):
  File "./test_mpx.py", line 24, in <module>
    tester = cantools.tester.Tester(dut, db, bus, bus_name=bus_name)
  File "/home/markrages/.local/share/virtualenvs/old_spampy-zmMsdUl_/lib/python3.8/site-packages/cantools/tester.py", line 285, in __init__
    self._messages[message.name] = Message(message,
  File "/home/markrages/.local/share/virtualenvs/old_spampy-zmMsdUl_/lib/python3.8/site-packages/cantools/tester.py", line 86, in __init__
    self.update(self._prepare_initial_signal_values())
  File "/home/markrages/.local/share/virtualenvs/old_spampy-zmMsdUl_/lib/python3.8/site-packages/cantools/tester.py", line 101, in update
    self._update_can_message()
  File "/home/markrages/.local/share/virtualenvs/old_spampy-zmMsdUl_/lib/python3.8/site-packages/cantools/tester.py", line 191, in _update_can_message
    data = self.database.encode(self.data,
  File "/home/markrages/.local/share/virtualenvs/old_spampy-zmMsdUl_/lib/python3.8/site-packages/cantools/database/can/message.py", line 938, in encode
    self.assert_signals_encodable(data, scaling=scaling)
  File "/home/markrages/.local/share/virtualenvs/old_spampy-zmMsdUl_/lib/python3.8/site-packages/cantools/database/can/message.py", line 655, in assert_signals_encodable
    used_signals = self.gather_signals(input_data)
  File "/home/markrages/.local/share/virtualenvs/old_spampy-zmMsdUl_/lib/python3.8/site-packages/cantools/database/can/message.py", line 560, in gather_signals
    raise EncodeError(f'A valid value for the multiplexer selector '
cantools.database.errors.EncodeError: A valid value for the multiplexer selector signal "S0" is required: Expected one of {1, 2, 3, 4 or 5}, but got 0
```
Are multiplexers supported in `cantools.tester()`?  Am I doing something wrong?

juleq, 2023-01-24:
Hi. Never used the tester. Apparently it initializes all signals to 0 if 0 is within a signals min/max. That is problematic in the case of your multiplexer because 0 is not a valid choice for it. Check the prepare inital [...] function to see whats happening.

Not sure what your use case does look like. If the tester implementation appears useful to you otherwise, you could consider proposing a fix. If not, just firing up a bunch of timers to update and send your messages should also be straightforward to implement test signal generators (if that is what you are after). I did it like this in js with the node-can package and it was a surprising low amount of code to do the whole residual bus simulation (but only sine waves between min and max, no specific behavioral implementations).

markrages, 2023-01-24:
My actual dbc file (which I can't share, alas) allows '0' in the mux, so I am facing a slightly different problem.  `Tester()` init throws all the alternative signals at the database encode at once, so the default strict checking throws exceptions.

I am able to workaround it with this monkeypatch:

```python
def __patched_update_can_message(self):
    """ Because https://github.com/cantools/cantools/issues/517 """
    arbitration_id = self.database.frame_id
    extended_id = self.database.is_extended_frame
    data = self.database.encode(self.data,
                                self.scaling,
                                self.padding,
                                strict=not self.database.is_multiplexed()) # <--- Added this line
    self._can_message = can.Message(arbitration_id=arbitration_id,
                                    is_extended_id=extended_id,
                                    data=data)
    
    if self._periodic_task is not None:
        self._periodic_task.modify_data(self._can_message)
        
cantools.tester.Message._update_can_message = __patched_update_can_message
```

juleq, 2023-01-24:
@andlaus Yes, the value table for the multiplexer signal does not contain 0. @markrages You could add 0 to the table in the dbc or, as @andlaus suggested, define a valid initial value. 1 is valid in your case. See the last line of your error trace.

markrages, 2023-01-26:
Is this the best way to find the allowed multiplexer values?

https://github.com/cantools/cantools/compare/master...markrages:cantools:tester_mux

PR #518 (2023-01-30): Allow cantools.tester to work with multiplexed signals
Resolves #517.
closed 2023-01-30 by juleq