cantools/cantools #524: BUS.fd_baudrate in dbc.py "_load_bus" function

BoningLiao, 2023-02-17:
We want to grab fb_baudrate through cantools in the dbc file, but the return value is None, trace the source code we found the _load_bus function doesn't check the fd_baudrate, and we add some code on our side, and it works.

andlaus, 2023-02-17:
yes, the DBC parser currently does not read the FD baudrate (only the [normal signalling rate](https://github.com/cantools/cantools/blob/master/cantools/database/can/formats/dbc.py#L1681-L1684); as far as I know, the only file format which currently reads the FD baudrate is ARXML). if you have code for that, we would be grateful if you contributed it...

timothyanderson096-ocdealcheck, 2026-10-21:
I checked the current DBC loader on master `3f407d8e76487208c376ec74411fb134bdf832c9`. `_load_bus()` reads DBName, Baudrate and the bus comment, but never populates `Bus.fd_baudrate`.

There is also a reproducible loss of FD-speed metadata using an existing fixture:

```python
import cantools

db = cantools.database.load_file("tests/files/arxml/system-4.2.arxml")
print(db.buses[0].baudrate, db.buses[0].fd_baudrate)
# 500000 2000000

loaded = cantools.database.load_string(db.as_dbc_string(), database_format="dbc")
print(loaded.buses[0].baudrate, loaded.buses[0].fd_baudrate)
# 500000 None
```

Could you share your working patch or a minimal DBC excerpt containing the relevant `BA_DEF_`, `BA_DEF_DEF_` and `BA_` entries for both rates? Please include the attribute's exact spelling, its units and whether the value comes from an explicit assignment or a default. The issue currently has no sample establishing the FD-rate attribute convention to support.

That would let us add focused load/round-trip regression coverage against a real file while preserving the existing nominal-bitrate behavior. This investigation does not yet implement a fix.

Investigated with AI assistance.

---
Timothy Anderson — Spotted Ridge Engineering / OC Labs · GDN (Global Decision Network)  
timothyanderson096@gmail.com · [Engineering evidence](https://github.com/timothyanderson096-ocdealcheck/gdn-engineering-benchmarks)

CYC7b, 2026-10-30:
I opened #877 to address this. It reads the FD data-phase rate from the Vector `BaudrateCANFD` database attribute (falling back to `BaudrateFD`) and writes `BaudrateCANFD` on dump, so ARXML → DBC → load now keeps `fd_baudrate` (500000 / 2000000 for `system-4.2.arxml`). Happy to adjust the attribute name if there's a different convention you'd prefer.

closed 2026-11-03 by andlaus