cantools/cantools #533: Read Multiplex ID Before Decoding

LucianoLim-Arcboats, 2023-03-10:
Hello, I'm wondering if there's a way to read the multiplex id of a given Message object before it is decoded?

andlaus, 2023-03-10:
Sorry, I don't really get what you mean here. can you please elaborate and also why you would want to do that?

if you mean a message's arbitration ID, this is provided by python-can or the code that reads the trace files. (this is not part of cantools, though.)

LucianoLim-Arcboats, 2023-03-10:
Sure, if I had a message like:
```
BO_ 2024 OBD2: 8 Vector__XXX
 SG_ S1_PID_0D_VehicleSpeed m13 : 31|8@0+ (1,0) [0|255] "km/h" Vector__XXX
 SG_ S1_PID_11_ThrottlePosition m17 : 31|8@0+ (0.39216,0) [0|100] "%" Vector__XXX
 SG_ S1 m1M : 23|8@0+ (1,0) [0|255] "" Vector__XXX
 SG_ Service M : 11|4@0+ (1,0) [0|15] "" Vector__XXX
```
I want to find the value of the signal `Service` in a given message without decoding the rest of the message for efficiency purposes.

andlaus, 2023-03-10:
probably the simplest solution is to make a new message which only contains the signals which you are interested in and use this to decode the data from the network

LucianoLim-Arcboats, 2023-03-10:
Ah I see. Thank you!

closed 2023-03-10 by LucianoLim-Arcboats