cantools/cantools #534: Signal.initial type discrepancy

zariiii9003, 2023-03-12:
Currently for dbc-files the `initial`-attribute of a `Signal` instance contains the raw integer value.
But for arxml it contains the physical or even a `NamedSignalValue`.

@andlaus What do you think about setting the `initial` attribute to the value of the `initial_int` in this line?
https://github.com/cantools/cantools/blob/df21fe3db7144e7e72e9882a661c5d4ea8c1dce7/cantools/database/can/formats/arxml/system_loader.py#L1485

andlaus, 2023-03-12:
> @andlaus What do you think about setting the initial attribute to the value of the initial_int in this line?

I think that cantools should use physical values for everything user-facing so that users never need to concern themselfs with the internal representation of values (basically everything that gets exposed by the 'Signal' class should be physical). Unfortunatly, it seems that this is currently inconsistent between the format parsers and we should switch all file formats to using physical values wherever possible, but this is going to be a bit tedious...

for DBCs: My presumption so far was that they specify the minima, maxima, etc using physical values but that -- if the scaling factors are all integers -- the type of these attributes can be `int`. (e.g. the  maximum of the `Temperature` signal of [`motohawk,dbc`](https://github.com/cantools/cantools/blob/master/tests/files/dbc/motohawk.dbc) is `270.47` Kelvins (which is a bit strange given that this is about -3°C and also does not really play well together with an offset of `250`...))

do you have any preference?

andlaus, 2023-03-12:
> so that users never need to concern themselfs with the internal representation of values 

basically I want this to work:
```python
m = mydb.get_message_by_name("MyNiceMessage")
sig_dict = {}
for signal in m.signals:
 sig_dict[signal.name] = signal.initial

print(m.encode(sig_dict))
```

This wouls be much harder if one needs to concern oneself with resolving enum values et al first...

zariiii9003, 2023-03-12:
Switching `initial` to a physical value might break a lot of code, even some cantools internal code (see tester).
I'd suggest to add another attribute like `scaled_initial` or something like that.

For your use case you could also write `m.encode(sig_dict, scaling=True)`

andlaus, 2023-03-13:
> Switching initial to a physical value might break a lot of code, even some cantools internal code (see tester).

I think that some of that code is already somewhat broken, as the ARXML loader has used physical values for quite a while now, and so far nobody seems to have noticed the inconsistency :/. (my impression from other issues over the last year or so is that probably a third to half of the users of cantools use ARXML...)

interestingly enough, the current tester code also seems to gets this confused: [`_prepare_initial_signal_values()`](https://github.com/cantools/cantools/blob/master/cantools/tester.py) prepares the initial physical signal values which ought to be used. But if `signal.initial` is not specified, it assigns the internal value of the minimum or 0 on lines [278](https://github.com/cantools/cantools/blob/master/cantools/tester.py#L278) and [282](https://github.com/cantools/cantools/blob/master/cantools/tester.py#L278). (besides this, the code might break if `signal.initial` is set but evaluates to `False`, e.g. by being `0.0`.)

> I'd suggest to add another attribute like scaled_initial or something like that.

good idea, but I would go for `internal_*` for the unscaled values...

> For your use case you could also write m.encode(sig_dict, scaling=False)

yes, but it is pretty cumbersome to mix-and-match signal values stemming from `decode()` (for which which you usually want to use the physical values) and the predetermined ones, e.g.
```python
can_msg = bus.recv()
db_msg = db.get_message_by_name("VehicleStatus")
msg_dict = db.decode(can_msg.arbitration_id, can_msg.data)
if msg_dict['Speed'] > 100:
  # set oil pump speed to maximum above 100 km/h
  msg_dict['OilPumpRPM''] =  db_msg.get_signal_by_name['OilPumpRPM'].maximum

can_msg.data = db.encode(msg_dict)
```
something like this gets pretty unintuitive to do if one has to manually juggle with internal vs physical representation of the values. I agree though, that we should provide access to the internal values via separate attributes...

zariiii9003, 2023-03-13:
So, to move this forward without breaking legacy code we could 

- replace the `initial` attribute with an `initial`-property, that keeps the current behavior
- raise a DeprecationWarning on access of the `initial` property
- introduce `raw_initial` and `scaled_initial` attributes (i don't really like `internal_*`, it sounds like it shouldn't be accessed)

The DeprecationWarning should point towards the new attributes.
If you agree i could draft a PR...

andlaus, 2023-03-13:
sounds good. we could also just decorate the `.initial` property with [`@deprecated`](https://pypi.org/project/deprecation/) instead of doing this manually, but maybe the additional dependency is not worth it. Further, I'd like to see the other such signal attributes (`.invalid`, `.minimum` and `.maximum`) handled the same way. agree?
 
(I especially like the fact that if the new attribute names are explicit enough so that if one of the file format loaders screws this up in the future, this is clearly a bug and the behavior can IMO be changed even if it potentially breaks downstream code...)

closed 2023-04-05 by zariiii9003

andlaus, 2023-04-06:
FYI: Today, I tried to make `signal.{maximum,minimum}` consistent with `signal.{initial,invalid}`. (As usual, it got more complicated than anticipated, PR will probably be ready sometime next week.) During this exercise, I noticed that all formats except ARXML specify these quantities using physical quantities and all formats expose physical quantities in these attributes. IOW: it is IMO a very good thing that #536 switched `signal.initial` for DBC to physical quantities as well! (we now only need to provided `signal.maximum_raw` and `signal.minimum_raw` as well...)

PR #536 (2023-04-05): Add raw_initial, scaled_initial and raw_invalid to Signal
Closes #534 

@andlaus Could you take a look at the failing test? I cannot find the initial value in the arxml file 🤔