cantools/cantools #539: How to parse SystemSignalLongSymbol

lautnerxing, 2023-03-28:
[image]
[image]

andlaus, 2023-03-28:
could you please elaborate? what is the issue? can you provide a complete file?

PUNITPNT, 2023-04-19:
hi ,
I also want to extract "SystemSignalLongSymbol",
Example:
Consider ->  SG_ signalname_0000 : 395|8@0+ (1,0) [0|255] ""  ABCD
SystemSignalLongSymbol" property is **set**  for  signalname_0000

BA_ "SystemSignalLongSymbol" SG_ xxxxx signalname_0000 "signalname_150000000000";

For a particular signal in dbc , if "SystemSignalLongSymbol" property is **set** 
       -> then Cantools is taking Signal Name as ""signalname_150000000000"" instead of  "signalname_0000"
       -> Cantools not storing "signalname_0000" this any where, when  "SystemSignalLongSymbol" property is **set** 

For a particular signal in dbc , if "SystemSignalLongSymbol" property is **NOT set** 
        -> then Cantools is taking Signal Name as ""signalname_0000""

geo4711, 2023-04-20:
Hi, the special attributes for long names are applied automatically. There might be some old tools that only support the short name (up to 32 chars). So there should be no need to separately "ask" for the long name. If you still want to do so for any reason, try to access the attribute "SystemSignalLongSymbol" as you would do for other dbc properties. As far as I remember, these special attributes are kept in the data structure, but for convenience the standard attributes for retrieving the name automatically reflect them, too.

PUNITPNT, 2023-04-20:
hi ,  

Can you help me to understand, why Can tools reading Signal name as value of SystemSignalLongSymbol" property . 
if this is property is **SET** for that signal

geo4711, 2023-04-21:
Hi,
this is due to the specification of the dbc file format. So cantools behaves like other common tools regarding that feature.
I guess that the very first versions of dbc file supported symbol names up to 32 chars only, and for compatibility reason this limitation has been kept, and the "workaround" with the special attributes has been introduced. All common tools (especially those from Vector) support long names of course, and the normal user does not need to care about that. I guess he won't even be aware of that concept (except he browses through the settings via CANdb++ and the integrated online help).