cantools/cantools #544: Support for DBC decoding pandas dataframes

MatinF, 2023-04-12:
Hi there,

We are interested in trying out this repo for one of our projects, which involves DBC decoding raw CAN bus data in the form of pandas dataframes. Is this possible with the repository? We were not able to find any examples of code achieving this.

andlaus, 2023-04-14:
I don't know what pandas dataframes are, but if you can get a sequence of payloads and the associated frame IDs, it basically comes down to this:

```python
import cantools

# [...]

db = cantools.database.load_file("my_nice_communication_description.dbc")

for frame_id, payload in unpacked_pandas_dataframes:
  print(db.decode_message(frame_id, payload)
```

andlaus, 2023-04-20:
Okay, I'm closing this issue for now. If the answer above does not work, don't hesitate to add further comments...

closed 2023-04-20 by andlaus