cantools/cantools #545: Decoding DBC "value table" data not generate enum

lysergicenemy, 2023-04-18:
When I create a table of values for some signal in the .DBC editor, the generated code looks like this:
```
Which is not very convenient for further use

It would be more correct, in my opinion, to create an enum for the signal to which the value table is assigned, so that the generated code looks like this:

juleq, 2023-04-18:
AFAIK it is at your compilers discretion to select a type for the enum here. That could lead to surprises when some other code makes assumptions on this structs layout (which naturally should be avoided anyway).

Could you explain the added convenience?

I think that only some static analyzers add the benefit of checking the type, compilers will eat a mismatch or maybe warn.

Using enums might add the requirement to add casts in some cases. With the defines, you could put the cast in the define to have proper typing.

These were just some thoughts on the matter, please note that I do not use the codegen ;).

lysergicenemy, 2023-04-18:
> AFAIK it is at your compilers discretion to select a type for the enum here. That could lead to surprises when some other code makes assumptions on this structs layout (which naturally should be [...]

For example, during debugging on the target platform, enums are automatically expanded into their definitions, which is very convenient because you do not need to remember or look at what the table means as if it were displayed in digital form

olowo726, 2025-01-22:
This functionality would be useful for me too due to similar reasons. In my case if there were an enum the corresponding number-to-text lookup would be automatically generated for the A2L file which would make it easier to debug the application. Is there any reason not to add this, perhaps as a switch so that the current functionality would be the default and the use can opt in for enums instead?

andlaus, 2025-01-24:
> Is there any reason not to add this

as @juleq mentioned, the issue with enums is that the size of enum objects depends on the whims of the compiler. That said, I think that the size of the frame structures currently does not necessarily match the size of the CAN frames observed on the bus anyway. Thus, if you or someone else implements that change to the code generator and add some test cases for it, I am willing to review and merge the corresponding PR (even if it comes with a new command line option for the code generator).

Ozuba, 2025-02-20:
#727 Hey i just tried to Implement this feature