cantools/cantools #551: Unknown frame id 217056256 (0xcf00400)

iszczyg, 2023-04-26:
Hello,
I try to decode can frames from Scania truck. I've dumped them with candump. Unfortunately, all frames are not recognized. I use j1939.db. So, I've decided to separete one frame:
(1682437707.894858) can0 0CF00400#F07D859B0FFFFF85
which is engine rpm and torque. It is properly recognized with my db file on asammdf. But, with cantools unfortunately I obtain message: Unknown frame id 217056256 (0xcf00400)
My command is:
cat scania.dat| python3 -m cantools decode  --single-line ./j1939.dbc 
The same for frames dumped from Toyota RAV4 and odb2.dbc file works perfectly. Should I use different command for extended frames? Please help me...

Best regards!

juleq, 2023-04-26:
That means that the id is not contained in your dbc. Since this is j1939, the issue usually is that the source address (least significant byte of the id) is different. Try finding that message in your dbc and compare its id to the one you try to decode.

iszczyg, 2023-04-26:
I have changed the frame id to 0CF004$FE (according to db file) and the response is:
(1682437707.894858) can0 0CF004FE#F07D859B0FFFFF85 ::
EEC1(
    EngTorqueMode: Low idle governor/no request (de,
    ActlEngPrcntTrqueHighResolution: 1.875 %,
    DriversDemandEngPercentTorque: 0 %,
    ActualEngPercentTorque: 8 %,
    EngSpeed: 499.375 rpm,
    SrcAddrssOfCtrllngDvcFrEngCntrl: 255,
    EngStarterMode: notAvailable,
    EngDemandPercentTorque: 8 %
)

But in j1939 description is, that it should be 0Cf00400. Probably by db is corrupted  :-(

juleq, 2023-04-26:
No, this is the way j1939 works. The id is also used to encode some protocol stuff (like the source address, SA).

The SA in the dbc is 0xfe while the one in your log is 0x00. 

Ideally one would use a tool that has proper j1939 support to cope with that. Cantools has some light support but I have never used that.

You can work around this by either changing the ids in your log to match your dbc or vice versa.

iszczyg, 2023-04-26:
Thanks a lot for your help!!!

juleq, 2023-04-26:
Welcome, been there ;).

geo4711, 2023-04-28:
The "source adresses" 0xFE and 0xFF are special values. They shall not be used for real ECUs. In a DBC, the meaning of these addresses are:
* 0xFF: broadcast (mainly used for the destination address, not for the source address)
* 0xFE: "default" or "any"
Defining a J1939 frame with SA/DA 0xFE means, that *any* SA/DA (from 0x00 to 0xFD) that this frame should be applied for any real message where the rest of the frame id matches (especially the PGN).
But as far as I remember, cantools does not support this "wildcard" feature.
You might observe similar problems with the priority in a j1939 frame. These are the first three bits in the 29 bit identifier. A common problem is having a frame definition like 0x18...... (which is prio 6), whereas your CAN trace might contain messages with 0x1C..... (which is prio 7). It is even allowed (for the ECU) to change the priority of a pgn dynamically. I could observe those problems with frames that have a variable length. The most common message "type" are DM1. They are either sent as single frames of as multiframe using the j1939 transport protocol. And the SAE itself proposes to use prio 6 for single DM1, but prio 7 for the multiframe usecase.
Most of the vector tools can deal with both "0xFE" and different prios.
But cantools, as juleq said, has a "light" implemtation for j1939.
Hope this helps and avoids some more suprise ;-)