cantools/cantools #553: Decoding overloaded messages

max3-2, 2023-05-03:
Hello,
so let me preface this by saying I am not sure if this is a case that is covered by CAN specs but I stubled across this and maybe there is a way to solve it.
I am sometimes using general "data windows" to find data on CAN where I just know the message bounds. So to break this down, I might have a message with a length of (in this example just) 2byte but just know that there is little endian data. So what I would do is the following in a dbc file:

```
BO_ 999 Template_0xFF: 2 Vector__XXX
 SG_ SigTest_8bit_0000_u : 0|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ SigTest_8bit_0000_s : 0|8@1- (1,0) [0|0] "" Vector__XXX
 SG_ SigTest_8bit_0001_u : 4|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ SigTest_8bit_0001_s : 4|8@1- (1,0) [0|0] "" Vector__XXX
 SG_ SigTest_8bit_0002_u : 8|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ SigTest_8bit_0002_s : 8|8@1- (1,0) [0|0] "" Vector__XXX

 SG_ SigTest_16bit_0000_u : 0|16@1+ (1,0) [0|0] "" Vector__XXX
 SG_ SigTest_16bit_0000_s : 0|16@1- (1,0) [0|0] "" Vector__XXX
```
this *overloads* the message since some of the signals occupy the same space. If I try to decode that using the `dbc` file, I'm getting errors from struct, namely that the message is too short. I traced that back to the way the unpacker is compiled which in this case would be something like `>u8s8u8s8u8s8u16s16` thus yielding 8 bytes for a 2 byte message.

This works without error (or at least no visible error) in some of the proprietary packages, e.g. MATLAB.

I had the idea of splitting this dbc in multiple messages with the same ID but since Im decoding coming from messages (e.g. calling `cantools.db.Database().decode_message(id, data)`) this is not unambigious. 
Is there a way to decode the message for each signal?

andlaus, 2023-05-03:
overlapping signals cannot be easily implemented due to some limitations of the bitstruct module (cf #205). As I see it, what would be needed for decoding multiple interpretations of the same data (or overlapping signals) is a "jump back" instruction for bitstruct, but it is unclear how encoding is supposed to work in that case...

max3-2, 2023-05-04:
Ah I See. I think I’ll try to build something around decode falling back to signal level if length does not match up and signals overlap. I’ll let you know if this works. 
Are there (new?) plans for a future implementation?

max3-2, 2023-06-26:
So I might have an idea - currently this is placed outside of the package but if someone with more experience could cross check I think an implementation is possbile (albeit slow during execution).

First, messages (signals) need a flag to show if they overlap. I have this implemented as follows but I suggest that during load (`Message` init) to add a new flag or change strict to add the information on message and / or signal basis, since the core check logic is already there. If they overlap, create a format that decodes the signal and not the whole message. If I understood the format geenration, this should add start padding if needed and thus allow decoding a signal anywhere on the message.

```python
from cantools.database.utils import (create_encode_decode_formats,
                                     decode_data as backend_decode)

# Add signal based decode formats
for message in dbc.messages:
            message_bits = 8 * message.length * [None]
            try:
                message._check_signal_tree(message_bits, message.signal_tree)
                message.has_overlap = False
            except CanError:
                logger.debug(f'Adding signal based format for {message}')
                message.has_overlap = True
                # Build signal based decoding
                for signal in message.signals:
                    dec_fmt = create_encode_decode_formats([signal], signal.length)
                    signal.dec_fmt = dec_fmt
```

Then, during decode, a check is performed. If the message has overlaps, signals are decoded on signal level using the compiled format for this signal. This looks somewhat like this:
```python
if dbc_msg.has_overlap:
        decoded = {}
        for signal in dbc_msg.signals:
            signal_decoded = backend_decode(
                data, signal.length, [signal], signal.dec_fmt,
                decode_choices, scaling, False)
            decoded |= signal_decoded

    else:
        decoded = dbc_db.decode_message(
            arbitration_id, data, decode_choices, scaling)
```

max3-2, 2023-06-26:
> As far as I can see, this will result in pretty bad decoding performance

Most certainly. But its an edge case anyway and increases support in those cases. I dont see speed being the major selling point.

> it _may_ not work because of the `bitstruct` module (which currently decodes all signals at once)

Can you further elaborate? My little testing shows it does decode the messages

andlaus, 2023-06-26:
>  I dont see speed being the major selling point.

I suppose you should have a conversation with @zariiii9003 ;)

> Can you further elaborate? My little testing shows it does decode the messages

yeah, it can work if the format passed to `bitstruct` is something like `p{signal.start}u{signal.length}` for every signal individually. besides this being rather inelegant, I don't think it is wise to drastically sacrifice performance for an edge case like this. What I would like to propose instead is to extend `bitstruct` by the concept of "negative padding" which sets back the cursor by a given number of bits: e.g. if this "negative padding" operation was represented using the format letter `q`, then interpreting a piece of data as 14 bit integer that overlaps its last four bits with a 10 bit integer could be achieved using the format `s14q4s10`. I'm not sure if @eerimoq is open to that idea, though.