cantools/cantools #562: pypy can not support cantools

Lit-fish-auto, 2023-05-12:
pypy3  successfully pip cantools
however,  import cantools, 
 error : can have no Listerner.

pypy not support cantools?

juleq, 2023-05-12:
There is a naming clash with some tool called "can" that people usually install by accident. Remove the can package, remove cantools, reinstall cantools.

Lit-fish-auto, 2023-05-13:
@juleq
uninstall  can and cantools,
inpypy ,   reinstall cantools, so there's only cantools,no can . but ,when it runs ,it warning  have no can mode .

in CPython, when install cantools, there's only cantools,no can. when it run, there is no wrong.

i think, when install  cantools ,  can should be installed in right version。because when install only  can , it installs can-0-0-0(version) in pypy, the version is wrong.

i don't know where is the fault, or how to pip the right version can (in pypy, the version can  needed by    cantools )?

Lit-fish-auto, 2023-05-13:
add

in CPython or pypy ,when  install cantools , python-can is automatically installed .

and then ,  
 in CPython, run it ,there's no warning.
in pypy ,  warning 1: no can ; warning 2: can have no listerner .

Lit-fish-auto, 2023-05-13:
python 3.9
pypy is the latest version   for python3.9

closed 2025-11-18 by ilanbiala