cantools/cantools #564: cantools decode 200million message

Lit-fish-auto, 2023-05-13:
CAN load  the asc or blf ,  about 200million message.

cantools decode these messages , and put the time ,signal name,physical value in dataframe.

overall, how to short the process time?

the way of pypy ,  multiprocessing is usefull ?

the numba do not support pandas and cantools ?

andlaus, 2023-05-13:
> overall, how to short the process time?

for such a large files, I would recommend biting the bullet and writing your program in C or C++: for BLF there is a C++ parser (http://tomilo.org/vector_blf.html) and using `cantools genererate_c_source`, you can generate the code to decode the frames...

Lit-fish-auto, 2023-05-14:
@andlus
can use vector_dbc  decode  data?
cantools generate c resource  can decode data  ?

andlaus, 2023-05-15:
> can use vector_dbc decode data?

no, but it can parse your BLF file, i.e. you get a sequence of frame IDs and payload data.

> cantools generate c resource can decode data ?

in conjunction with your favourite C compiler you can use it to decode above sequence of can frames...

Lit-fish-auto, 2023-05-15:
@andlaus
cantools generate c resource  can be C++ used?
above all , thank a lot.

andlaus, 2023-05-16:
It generates C code, but C is a subset of C++. So yes, it generates C++ code.

andlaus, 2023-05-16:
you have to use the `cantools` command on the command line. That said, if you're not already at least somewhat proficient in C/C++, I would advise you not to waste your time and hire some consultant to do the job for you...