cantools/cantools #566: Metis CAN dbc and sym, causes errors when read and when generating C

guy-radford-sunswap, 2023-05-16:
This issue comes off the back of Issue #249 

We are using the Metis Battery Safety CAN Sensor, they provide CAN dbc and sym file, neither of which can be read by `cantools`
https://www.metisengineering.com/store/p13/R%26D_Battery_Safety_CAN_Sensor.html

**`cantools generate_c_source BSS_V7.dbc`**
```
error: DBC: "unsupported operand type(s) for *: 'decimal.Decimal' and 'float'"
```

**`cantools generate_c_source BSS_V7.sym`**
```
error: SYM: "Invalid syntax at line 118, column 24: "Sig=CANStartAddress 32 >>!<<-h""
```

After removing the two instances of `-h` from the `BSS_V7.sym` file, the file would load and output the `c` files. 
However, they will not compile (referencing back to the issue #249 )

**`gcc -Wall -c bss_v7.c`**
```
In file included from bss_v7.c:33:
./bss_v7.h:330:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:337:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:344:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:351:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:358:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:365:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:372:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:379:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:386:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:393:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:400:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:407:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:414:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:421:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:428:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:435:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:442:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:449:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
./bss_v7.h:456:14: error: duplicate member 'unique_id'
    uint32_t unique_id;
             ^
./bss_v7.h:323:14: note: previous declaration is here
    uint32_t unique_id;
             ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
```

andlaus, 2023-05-16:
hm, the error with the DBC goes away with the branch of #555. further, with the #555 branch, the C output compiles fine on my machine:
```
> cantools generate_c_source BSS_V7.dbc 
Successfully generated ./bss_v7.h and ./bss_v7.c.
> gcc -c bss_v7.c 
>
```

@guy-radford-sunswap: as a quick fix, I recommend the branch of #555 and the DBC until we figure this out.

 @zariiii9003: a reason to remove the decimal scaling stuff?

andlaus, 2023-05-16:
depending on your point of view, the problem with the sym file seems to either be an issue with the file itself, or a limitation of the code generator: since the `UniqueID` signal is in all multiplexed parts of the `BSS_Config` message, it leads to name conflicts with the `{db_name}_{message_name}_{signal_name}` scheme. (we would need something like `{db_name}_{message_name}_{multiplexer_id}_{signal_name}`. Or, in this particular case, the `UniqueID` signal should be moved to the static part of the message...)