cantools/cantools #568: Comma ai's opendbc dbc files not working

ashtyn3, 2023-05-20:
Hello,

To preface this whole issue, I am very new to this CAN and car hacking world.

I've recently been trying to use cantools to read can messages from cangen. But for some reason when I try to use a dbc file from comma ai's opendbc (specifically [link to file](https://raw.githubusercontent.com/commaai/opendbc/master/honda_civic_ex_2022_can_generated.dbc)) I get the following error: `error: DBC: "Invalid syntax at line 171, column 13: "CM_ SG_ 304 >>!<<"Seems to be platform-agnostic";""`.  I tried just deleting the lines but it only got worse from there. 

Is this a known bug? If so is there any solutions?

juleq, 2023-05-20:
Had a look at the dbc file. It is not in Vector dbc format. I assume comma has their own tooling that is loosely based on dbc?

I noticed at first glance:
- preamble missing
- order not consistent with Vector tooling

And, according to your report, comment syntax also appears off.

A good test is to try to open the file with Vectors CANdb++ program. It will probably just due in a fire, though.

ashtyn3, 2023-05-20:
Any recommendations? Is there another good source for DBC files?

juleq, 2023-05-20:
I usually get mine from vendors.

What you could do: Take a blank dbc file (e.g. from the tests in cantools) and copy over BO (messages) and SG (signals) of interest using a text editor. But the dbc file format is not very user friendly to hand- edit...