cantools/cantools #576: How can I decode ethernet data which is logged by CANoe?

linpanusst, 2023-06-07:
Hi, Miss/Mr. Recently, I work with some data which is logger from CANoe. And its database is arxml. I use 
`import can
data = can.BLFReader(xxx)`
But the output show that, I can't read the ethernet data in the blf. Even the bytes data didn't appear.
Do you have any plan to update this function?
If you need any sample data. I can provide it to you via email.

HansSchriefersPorsche, 2023-11-16:
Hi @linpanusst, did you get an update here? We're having a similiar problem, but concerned, that this package will not do the job.

zariiii9003, 2023-11-16:
You're a bit lost. The BLFReader is part of the python-can library, not cantools. But even there Ethernet is out of scope for the *CAN* library. So you'd have to implement it yourself and submit a well documented and tested PR, if you want that feature.