cantools/cantools #581: DBC Decode Message In Error

MichaelBMiner, 2023-06-13:
Hello,

I am using DBCPPP to generate CAN data on an embedded linux device. I am sending this CAN data to a Raspberry Pi running cantools. 

In my application I can use DBCPPP to encode and decode the message no problem. When cantools receives the message, the decoded data is all wrong. My device is filling and sending PGN 

```
BO_ 2566568016 PGN_64184_EVSE_1_DC_Status_1: 8 EVCC
 SG_ EVSE1_DC_Charging_State : 0|4@1+ (1,0) [0|15] "" EV
 SG_ EVSE1_Isolation_Status : 4|4@1+ (1,0) [0|15] "" EV
 SG_ EVSE1_Present_DC_Charging_Voltage : 8|16@1+ (0.05,0) [0|3212.75] "V" EV
 SG_ EVSE1_Present_DC_Charging_Current : 24|16@1- (0.05,-1600) [-1600|1612.75] "A" EV
 SG_ EVSE1_Voltage_Limit_Achieved : 40|2@1+ (1,0) [0|3] "" EV
 SG_ EVSE1_Current_Limit_Achieved : 42|2@1+ (1,0) [0|3] "" EV
 SG_ EVSE1_Power_Limit_Achieved : 44|2@1+ (1,0) [0|3] "" EV
 SG_ EVSE1_Processing_State : 46|2@1+ (1,0) [0|3] "" EV
 SG_ EVSE1_Status : 48|4@1+ (1,0) [0|15] "" EV
 SG_ EVSE1_Response_Code : 56|8@1+ (1,0) [0|255] "" EV
```

My CAN frame data field contains the following 0x000000B036D08400, I can confirm the raspberry pi sees the same data. This data is to fill the above PGN with values EVSE1_Present_DC_Charging_Voltage = 700V and EVSE1_Present_DC_Charging_Current 100A. 

Instead I get
'EVSE1_Present_DC_Charging_Voltage': 2054.65
'EVSE1_Present_DC_Charging_Current': -1597.3

The two devices share a DBC file so they can properly encode/decode messages. My python code is not complicated. I manipulate the DBC ID and CAN ID internally on both devices, but the tools can always match an ID in the DBC file. 

`data = self.dbc.decode_message(ext_can_id_with_our_sa, message.data)`

andlaus, 2023-06-13:
this seems to be a mess-up with the byte/bit order between the devices. since I'm pretty sure that the python part of cantools handles this correctly, how is the data encoded on the embedded device?

MichaelBMiner, 2023-06-13:
To encode the data the application does the following

```
        static const dbcppp_Network* net;
	static uint32_t num_unique_dbc_ids;
	net = dbcppp_NetworkLoadDBCFromFile(DEFAULT_DBC_FILE_LOCATION);
	num_unique_dbc_ids = dbcppp_NetworkMessages_Size(net);
	for (uint32_t i = 0; i < num_unique_dbc_ids; i++) {
		const dbcppp_Message* msg = dbcppp_NetworkMessages_Get(net, i);
		uint64_t msg_id = dbcppp_MessageId(msg);
		if(to_send.can_id == msg_id) {
			for(uint32_t j=0; j<dbcppp_MessageSignals_Size(msg); j++) {
				const dbcppp_Signal* sig = dbcppp_MessageSignals_Get(msg, j);
				printf("Parsing SPN %s\n", dbcppp_SignalName(sig));
				uint64_t temp_raw = 0;
				uint64_t raw = 0; //dbcppp_SignalPhysToRaw(sig, dbc_id_info_array[index].spns[i].phys);
				if(strcmp("EVSE1_Present_DC_Charging_Voltage", dbcppp_SignalName(sig)) == 0) {
					raw = dbcppp_SignalPhysToRaw(sig, voltage);
				}
				if(strcmp("EVSE1_Present_DC_Charging_Current", dbcppp_SignalName(sig)) == 0) {
					raw = dbcppp_SignalPhysToRaw(sig, current);
				}
				dbcppp_SignalEncode(sig, raw, &temp_raw);
				uint64_t bitsize = dbcppp_SignalBitSize(sig);
				raw_value |= (temp_raw << bitsize);
			}
		}
	}

	memcpy(to_send.data, &raw_value, sizeof(raw_value));
	printf("temp_raw: %" PRIu64 "\n", raw_value);
```

After this the CAN frame is transmitted.

andlaus, 2023-06-13:
does this produce the same data as when encoded with cantools using [`message.encode()`](https://github.com/cantools/cantools/blob/master/cantools/database/can/message.py#L884). If it isn't the embedded code is most likely wrong...

juleq, 2023-06-13:
How does the embedded device decode? Byte by byte or does it use wider types like u32 or u64? If the latter, endianness could catch you by surprise.

MichaelBMiner, 2023-06-13:
To decode I run the code

```
for (uint32_t i = 0; i < num_unique_dbc_ids; i++) {
		const dbcppp_Message* msg = dbcppp_NetworkMessages_Get(net, i);
		uint64_t msg_id = dbcppp_MessageId(msg);
		if(to_send.can_id == msg_id) {
			for(uint32_t j=0; j<dbcppp_MessageSignals_Size(msg); j++) {
				const dbcppp_Signal* sig = dbcppp_MessageSignals_Get(msg, j);
				printf("Parsing SPN %s\n", dbcppp_SignalName(sig));
				shift_size += dbcppp_SignalBitSize(sig);
				uint64_t mask = (1ULL << dbcppp_SignalBitSize(sig)) - 1;
				uint64_t temp_raw = (raw_value >> shift_size) & mask;
				uint64_t phys = (uint64_t)dbcppp_SignalRawToPhys(sig, temp_raw);
				uint32_t start_bit = dbcppp_SignalStartBit(sig);
				printf("temp_raw: %" PRIu64 ", phys: %" PRIu64 ", start_bit: %u\n", temp_raw, phys, start_bit);
			}
		}
	}
```

juleq, 2023-06-13:
What is the endianness of the machine? Which MCU? You could check if DBCPPP has endianness agnostic code.

If it is not this, not getting the IDs right would also explain.

MichaelBMiner, 2023-06-13:
I am using a scheduler to schedule my TX messages. This is how I populate the data for this PGN.

```
    def send_test(battery, scheduler, time, priority):
    battery.data_dict = {}
    msg = CanHandler.getInstance().find_message_by_name("PGN_64184_EVSE_1_DC_Status_1")
    if msg is not None:
        battery.data_dict["EVSE1_DC_Charging_State"] = 0
        battery.data_dict["EVSE1_Isolation_Status"] = 0
        battery.data_dict["EVSE1_Present_DC_Charging_Voltage"] = 700
        battery.data_dict["EVSE1_Present_DC_Charging_Current"] = 100
        battery.data_dict["EVSE1_Voltage_Limit_Achieved"] = 0
        battery.data_dict["EVSE1_Current_Limit_Achieved"] = 0
        battery.data_dict["EVSE1_Power_Limit_Achieved"] = 0
        battery.data_dict["EVSE1_Processing_State"] = 0
        battery.data_dict["EVSE1_Status"] = 0
        battery.data_dict["EVSE1_Response_Code"] = 0
        CanHandler.getInstance().send_can_frame(battery, msg.frame_id)
    scheduler.enter(time, priority, send_test, (battery, scheduler, time, priority))
```

In my CAN handling code I have a send function

```
 def send_can_frame(self, battery, frame_id):
        try:
            ext_can_id_without_sa = (frame_id & REMOVE_SOURCE_ADDR_MASK)
            ext_can_id_with_our_sa = (ext_can_id_without_sa | self.sa)
            dbc_can_id = (4 << 29) | frame_id

            data = self.dbc.encode_message(ext_can_id_with_our_sa, battery.data_dict)
            message = can.Message(arbitration_id=ext_can_id_with_our_sa, data=data)
            self.can_tx.send(message)
        except Exception as err:
            print("ERROR {} {}".format(battery.data_dict, err))
```

This is returning the error 

ERROR {'EVSE1_DC_Charging_State': 0, 'EVSE1_Isolation_Status': 0, 'EVSE1_Present_DC_Charging_Voltage': 700, 'EVSE1_Present_DC_Charging_Current': 100, 'EVSE1_Voltage_Limit_Achieved': 0, 'EVSE1_Current_Limit_Achieved': 0, 'EVSE1_Power_Limit_Achieved': 0, 'EVSE1_Processing_State': 0, 'EVSE1_Status': 0, 'EVSE1_Response_Code': 0} Signed integer value 34000 out of range.

34000 is the raw value for a current of 100A. (100 - (-1600))/0.05 = 34000.

andlaus, 2023-06-13:
hm, this does not really look like code that uses cantools (at least not a somewhat contemporary version): e.g, there is no `.find_message_by_name()` function (it is [`get_message_by_name()`](https://github.com/cantools/cantools/blob/master/cantools/database/can/database.py#L375))

> 34000 is the raw value for a current of 100A. (100 - (-1600))/0.05 = 34000.

I'm pretty sure that you're not using cantools, since the string "out of range" does not occur in the codebase:
```bash
~/src/cantools > find cantools -iname "*.py" | xargs grep -i "out of range"
~/src/cantools >
```

andlaus, 2023-06-13:
I suppose I cannot really help you with this. To get the binary data which cantools would send, you can use
```python
import cantools

db = cantools.database.load_file("my_dbc_file.dbc")
signal_dict ={ 
    "EVSE1_DC_Charging_State": 0,
    "EVSE1_Isolation_Status": 0,
    "EVSE1_Present_DC_Charging_Voltage": 700,
    "EVSE1_Present_DC_Charging_Current": 100,
    "EVSE1_Voltage_Limit_Achieved": 0,
    "EVSE1_Current_Limit_Achieved": 0,
    "EVSE1_Power_Limit_Achieved": 0,
    "EVSE1_Processing_State": 0,
    "EVSE1_Status": 0,
    "EVSE1_Response_Code": 0,
}

msg = db.get_message_by_name("PGN_64184_EVSE_1_DC_Status_1")

print(msg.encode(signal_dict).hex())
```

(modulo minor mistakes, I haven't run the code above)

MichaelBMiner, 2023-06-13:
I have taken your code directly. 

```
     try:
        dbc = cantools.database.load_file(os.path.abspath(os.path.join(os.path.dirname( __file__ ), '../', ''))+ config['DBC']['DBCPath'])

        print(dbc.messages)

        signal_dict ={ 
            "EVSE1_DC_Charging_State": 0,
            "EVSE1_Isolation_Status": 0,
            "EVSE1_Present_DC_Charging_Voltage": 700,
            "EVSE1_Present_DC_Charging_Current": 100,
            "EVSE1_Voltage_Limit_Achieved": 0,
            "EVSE1_Current_Limit_Achieved": 0,
            "EVSE1_Power_Limit_Achieved": 0,
            "EVSE1_Processing_State": 0,
            "EVSE1_Status": 0,
            "EVSE1_Response_Code": 0,
        }

        msg = dbc.get_message_by_name("PGN_64184_EVSE_1_DC_Status_1")
        data = msg.encode(signal_dict).hex()

        print(data)
    except Exception as err:
        print("ERROR {} {}\n".format(signal_dict, err))
```

When running msg.encode I get the error 

ERROR {'EVSE1_DC_Charging_State': 0, 'EVSE1_Isolation_Status': 0, 'EVSE1_Present_DC_Charging_Voltage': 700, 'EVSE1_Present_DC_Charging_Current': 100, 'EVSE1_Voltage_Limit_Achieved': 0, 'EVSE1_Current_Limit_Achieved': 0, 'EVSE1_Power_Limit_Achieved': 0, 'EVSE1_Processing_State': 0, 'EVSE1_Status': 0, 'EVSE1_Response_Code': 0} Signed integer value 34000 out of range.

MichaelBMiner, 2023-06-13:
Did some more digging in dbcppp for the data

signal_dict ={ 
        "EVSE1_DC_Charging_State": 0,
        "EVSE1_Isolation_Status": 0,
        "EVSE1_Present_DC_Charging_Voltage": 700,
        "EVSE1_Present_DC_Charging_Current": 30,
        "EVSE1_Voltage_Limit_Achieved": 0,
        "EVSE1_Current_Limit_Achieved": 0,
        "EVSE1_Power_Limit_Achieved": 0,
        "EVSE1_Processing_State": 0,
        "EVSE1_Status": 0,
        "EVSE1_Response_Code": 0,
    }

I get a hex value of 0x7f5836b0000000
The same data in cantools gives 00b036587f000000

MichaelBMiner, 2023-06-13:
If I change my PGN to remove the signed data 

```
BO_ 2566568000 PGN_64184_EVSE_1_DC_Status_1: 8 EV
 SG_ EVSE1_DC_Charging_State : 0|4@1+ (1,0) [0|15] "" EVCC
 SG_ EVSE1_Isolation_Status : 4|4@1+ (1,0) [0|15] "" EVCC
 SG_ EVSE1_Present_DC_Charging_Voltage : 8|16@1+ (0.05,0) [0|3212.75] "V" EVCC
 SG_ EVSE1_Present_DC_Charging_Current : 24|16@1+ (0.05,0) [0|1612.75] "A" EVCC
 SG_ EVSE1_Voltage_Limit_Achieved : 40|2@1+ (1,0) [0|3] "" EVCC
 SG_ EVSE1_Current_Limit_Achieved : 42|2@1+ (1,0) [0|3] "" EVCC
 SG_ EVSE1_Power_Limit_Achieved : 44|2@1+ (1,0) [0|3] "" EVCC
 SG_ EVSE1_Processing_State : 46|2@1+ (1,0) [0|3] "" EVCC
 SG_ EVSE1_Status : 48|4@1+ (1,0) [0|15] "" EVCC
 SG_ EVSE1_Response_Code : 56|8@1+ (1,0) [0|255] "" EVCC
```

Now I can set my current to any valid value and my hex output data is what I expect, but in a different order.

andlaus, 2023-06-13:
hm, maybe this is an exception raised by the `bitstruct` module. this indicates that the `EVSE1_Present_DC_Charging_Current` has to few bits to represent a signed integer value of 34000 (it must be at least 17 bits long).

MichaelBMiner, 2023-06-14:
My question then is, how do I proceed? Why is the voltage able to be properly converted from 700 to 14000? Why when I remove the negative offset and minimum value does it start to work?

MichaelBMiner, 2023-06-14:
Here is my original PGN entry in the DBC file

BO_ 2566568016 PGN_64184_EVSE_1_DC_Status_1: 8 EVCC
 SG_ EVSE1_DC_Charging_State : 0|4@1+ (1,0) [0|15] "" EV
 SG_ EVSE1_Isolation_Status : 4|4@1+ (1,0) [0|15] "" EV
 SG_ EVSE1_Present_DC_Charging_Voltage : 8|16@1+ (0.05,0) [0|3212.75] "V" EV
 SG_ EVSE1_Present_DC_Charging_Current : 24|16@1- (0.05,-1600) [-1600|1612.75] "A" EV
 SG_ EVSE1_Voltage_Limit_Achieved : 40|2@1+ (1,0) [0|3] "" EV
 SG_ EVSE1_Current_Limit_Achieved : 42|2@1+ (1,0) [0|3] "" EV
 SG_ EVSE1_Power_Limit_Achieved : 44|2@1+ (1,0) [0|3] "" EV
 SG_ EVSE1_Processing_State : 46|2@1+ (1,0) [0|3] "" EV
 SG_ EVSE1_Status : 48|4@1+ (1,0) [0|15] "" EV
 SG_ EVSE1_Response_Code : 56|8@1+ (1,0) [0|255] "" EV

I have changed it to 

BO_ 2566568016 PGN_64184_EVSE_1_DC_Status_1: 8 EVCC
 SG_ EVSE1_DC_Charging_State : 0|4@1+ (1,0) [0|15] "" EV
 SG_ EVSE1_Isolation_Status : 4|4@1+ (1,0) [0|15] "" EV
 SG_ EVSE1_Present_DC_Charging_Voltage : 8|16@1+ (0.05,0) [0|3212.75] "V" EV
 SG_ EVSE1_Present_DC_Charging_Current : 24|16@1+ (0.05,-1600) [-1600|1612.75] "A" EV
 SG_ EVSE1_Voltage_Limit_Achieved : 40|2@1+ (1,0) [0|3] "" EV
 SG_ EVSE1_Current_Limit_Achieved : 42|2@1+ (1,0) [0|3] "" EV
 SG_ EVSE1_Power_Limit_Achieved : 44|2@1+ (1,0) [0|3] "" EV
 SG_ EVSE1_Processing_State : 46|2@1+ (1,0) [0|3] "" EV
 SG_ EVSE1_Status : 48|4@1+ (1,0) [0|15] "" EV
 SG_ EVSE1_Response_Code : 56|8@1+ (1,0) [0|255] "" EV

The endianness of the EVSE1_Present_DC_Charging_Current  signal was the issue.

closed 2023-06-14 by MichaelBMiner

juleq, 2023-06-14:
Thanks for letting us know. I do not know what the Motorola people were smoking, but there is only one valid endianness. Someone should have told them :).