cantools/cantools #584: 39.0.0rc1: cannot import `cantools` due to circular reference

mon, 2023-06-27:
Not quite sure what change led to this, but it works fine on 38. Installed rc1 via pip.

```
Python 3.11.0 (main, Oct 24 2022, 18:26:48) [MSC v.1933 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import cantools
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\mon\AppData\Local\Programs\Python\Python311\Lib\site-packages\cantools\__init__.py", line 13, in <module>
    from . import database as db  # isort: skip
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: cannot import name 'database' from partially initialized module 'cantools' (most likely due to a circular import) (C:\Users\mon\AppData\Local\Programs\Python\Python311\Lib\site-packages\cantools\__init__.py)
```

andlaus, 2023-06-27:
very weird. does it work with the git version? (i.e., maybe the issue is caused by the pypi package...)

BTW: we should mention your linter in the README before the final release. can you open a PR to that end?

mon, 2023-06-27:
Indeed, installing the git version (via `pip install -U .`) works fine.
```
>>> import cantools
>>> cantools.__version__
'39.0.0-rc1'
```

> we should mention your linter in the README before the final release

I'm not quite sure where's the best spot to link it! Maybe nearby the `Other C code generators:` section?

mon, 2023-06-27:
Actually, I spoke too soon, since I ran my test terminal in the cantools git repo.... It doesn't work, but a different error!

Site packages dir shortened to `....` for brevity

```
Python 3.11.0 (main, Oct 24 2022, 18:26:48) [MSC v.1933 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import cantools
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "....\cantools\__init__.py", line 13, in <module>
    from . import database as db  # isort: skip
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "....\cantools\database\__init__.py", line 8, in <module>
    from . import can, diagnostics, utils
  File "....\cantools\database\can\__init__.py", line 2, in <module>
    from .database import Database
  File "....\cantools\database\can\database.py", line 23, in <module>
    from .formats import arxml, dbc, kcd, sym
  File "....\cantools\database\can\formats\arxml\__init__.py", line 6, in <module>
    from ...internal_database import InternalDatabase
  File "....\cantools\database\can\internal_database.py", line 8, in <module>
    from .message import Message
  File "....\cantools\database\can\message.py", line 42, in <module>
    from .signal import Signal
  File "....\cantools\database\can\signal.py", line 5, in <module>
    from ..conversion import BaseConversion, IdentityConversion
ModuleNotFoundError: No module named 'cantools.database.conversion'
>>>
```

zariiii9003, 2023-06-27:
Sub packages are missing because of #583. That's my fault, i'll fix that.
@andlaus could you yank that release?

closed 2023-06-27 by zariiii9003

andlaus, 2023-06-27:
> Maybe nearby the `Other C code generators:` section?

possibly, we can introduce a "Open Source Projects Using cantools" (or similar) section?

andlaus, 2023-06-27:
> @andlaus could you yank that release?

I don't have the PyPI password, but maybe @juleq can help out here? I'll tag a RC2 prerelease, though...

juleq, 2023-06-27:
@andlaus Yank rc1?

PR #585 (2023-06-27): Fix ImportError
Closes #584 

Maybe we should switch to a `src/cantools` package structure in the future. The tests would have failed in this case.