cantools/cantools #588: 39.0.0rc2: Messages using default GenMsgCycleTime get explicitly serialised

mon, 2023-06-28:
I'm not sure if this was caused by the `Decimal` removal or something else - my DBC specifies a default `GenMsgCycleTime` of 100, but any message using this value now gets an explicit `BA_` entry in the saved dbc.

andlaus, 2023-06-30:
weird: I would be *very* surprised if this was caused by the `Decimal` removal. can you bisect the issue?

mon, 2023-06-30:
Thanks for the bump, I was being a bit lazy by not bisecting this myself. As it turns out, it's still a bug, just not the one I thought.
 This goes back to at or before 37.1.1, the oldest release my formatter runs on.

I have another script that turns a non-standard "txPeriod" attribute into "GenMsgCycleTime", so I can utilise the `cycle_time` signal attribute. On the first run, this works fine, but the _second_ run it inserts all the default values (I never ran the script twice, so didn't notice until now 😅).

The culprit is `get_cycle_time` falling back to the default (which is sane behaviour) but the serialisation not performing a "only serialise if not default" step.

If you're OK with me changing the behaviour, I can make a PR on Monday.
Though, should this apply to all attributes? It certainly is redundant information, but I'm not sure if some people rely on intentionally serialising some defaults.

andlaus, 2023-06-30:
> If you're OK with me changing the behaviour, I can make a PR on Monday.

this feature seems legit to me  (albeit pretty low-level): IMO, we should strive to be as invariant as possible with regard to load-store cycles, so that PR would certainly be appreciated..

> Though, should this apply to all attributes? It certainly is redundant information, but I'm not sure if some people rely on intentionally serialising some defaults.

I think it would be nice it dumped files used the same approach regarding attribute defaults as the files from which they were loaded. I don't think we should be too dogmatic about this, though...

mon, 2023-07-01:
> invariant as possible with regard to load-store cycles

In that case, I think it makes sense to match existing attribute behaviour, i.e.
- If a cycle time was loaded OR set at runtime, even if it's copying the default, it will be stored
- if a cycle time wasn't loaded, it won't be stored

mon, 2023-07-03:
I think it will actually be easier to make `_cycle_time`, if None, fallback to to the DbcSpecifics instead of automatically setting it at load time. This will let the behaviour match what I wrote above. Looking at the other parsers that use it (kcd and sym), they don't have any concept of a database-level default, so it won't affect them. It'll also close #540 (only create the top-level definition if any messages have been given cycle times). Pr will be coming in a bit.

PR #592 (2023-08-03): Make cycle_time serialisation more consistent for DBC files
- Moves default fallback to the getter instead of the initial load from DBC
- Only creates the `GenMsgCycleTime` attribute if it's missing _and_ needed
- Doesn't serialise default values if they haven't been changed

It _feels_ like the current test suite is sufficient to validate this change, but feel free to request some more.

Closes #588, closes #540.
closed 2023-08-03 by zariiii9003