cantools/cantools #591: Decode multiplexed signal behaves strange. What am I doing wrong?

LukeSkywalker92, 2023-06-29:
Hey,

I am somehow a newbe to Can, so this might be a stupid question.
My Problem is, that when I try to decode a multiplexed signal, the first value of the returned dict, which carries the multiplexer id, has the name of the first multiplexer as a key. Should this really be the case? Shouldn't the key be something generic like `MUX`?

Please see the following example for better understanding my Problem:

When I use for example the following `.sym' file:

```
FormatVersion=6.0 // Do not edit this line!
Title="CanMsgs"

{ENUMS}

{SIGNALS}

{SENDRECEIVE}
[TestSig]
ID=700h
Len=2
CycleTime=1000
Mux=FSM 0,8 1
Var=alert1 unsigned 8,1
Var=alert2 unsigned 9,1
Var=alert3 unsigned 10,1

[TestSig]
Len=2
CycleTime=1000
Mux=DCM 0,8 2
Var=alert1 bit 8,1
Var=alert2 bit 9,1
Var=alert3 unsigned 10,1
```

And I run the following code:

```python
import cantools

db = cantools.database.load_file('test.sym')

print(db.decode_message('TestSig', b'\x01\x00\x00\x00\x00\x00\x00\x00'))
print(db.decode_message('TestSig', b'\x02\x00\x00\x00\x00\x00\x00\x00'))
```

I get the following output:

```bash
{'FSM': 1, 'alert1': 0, 'alert2': 0, 'alert3': 0}
{'FSM': 2, 'alert1': 0, 'alert2': 0, 'alert3': 0}
```

So as you see, the first key is always `FSM`. If I want to probe on the multiplexer id, I can of course ignore the key and just take the value of the first entry, but this somehow feels not like a proper way of doing it.

Thank you very much for your help in advance.

Cheers
Luke

andlaus, 2023-06-30:
in python, ordinary dictionaries do not have an ordering, so there is no "first" entry, because -- in principle -- the order of the keys is undefined. what you probably need is to access the values of the dictionary by name, i.e. something like:
```python
# [...]
decoded_signals = db.decode_message('TestSig', b'\x01\x00\x00\x00\x00\x00\x00\x00')
if decoded_signals['FSM'] == 1:
    # do something if the selector signal was set to 1
    print("Selected 1")
elif  decoded_signals['FSM'] == 2:
    # do something if the selector signal was set to 1
    print("Selected 2")
```

(you probably want to assign different signals to a given mux value, else using a multiplexer does not make much sense.)

LukeSkywalker92, 2023-06-30:
Hey,
first of all, thank you for your answer.

What you describe about ordering is actually my concern.
But what I don't understand is why the key for the multiplexer value is always the first multiplexer value defined in the `.sym` file. In my example I specify `FSM` for multiplexer 1 and `DCM` for the second. But the key in the decoded dict is always `FSM`....

juleq, 2023-06-30:
Hi. cantools started with dbc support. Dbcs do not know the concept of naming multiple multiplexers directly. In dbc, the multiplexor is treated as a signal with just one name. It just is marked to be a multiplexor. The signals are then multiplexed signals.

I think this is the reason cantools cannot reflect the naming options that sym offers there.

LukeSkywalker92, 2023-06-30:
Thank you for clarification. Then I need to find a workaround for what I have planned 😄

closed 2023-06-30 by LukeSkywalker92