cantools/cantools #597: Using Cantools trying to generate a .dbc file

Azad202222, 2023-07-24:
Traceback (most recent call last):
  File "D:\Python_Codes\venv\lib\site-packages\cantools\database\can\formats\dbc.py", line 1737, in try_remove_attribute
    dbc.attributes.pop(name)
TypeError: set.pop() takes no arguments (1 given)

Need help in trying to solve this issue.
dump_file(db, output_file)
After calling dump_file this issue is coming

andlaus, 2023-07-24:
are you sure that you use an unmodified version of cantools? your `message.dbc.attributes` is a set, but looking at the DBC loader code, I cannot see how this could happen. (it is supposed to be an `OrderedDict`.) If you do indeed use an unmodified version, can you boil down your source DBC file to something non-confidential where this still happens?

Azad202222, 2023-07-24:
[image]
Not using unmodified version... Just installed the packages in Pycharm

andlaus, 2023-07-24:
can you provide a non-confidential version of your source DBC file that exhibits this issue?

Azad202222, 2023-07-24:
Im not using source DBC... Getting all the datas by importing from an excel sheet... And mapping those values to the db object... And generating a DBC file.

andlaus, 2023-07-24:
> Im not using source DBC... Getting all the datas by importing from an excel sheet... And mapping those values to the db object... And generating a DBC file. The DBC file only not getting generated.. Throwing error

then this code is wrong: you made the `message.dbc.attributes` object a `set`, but it is supposed to be an `OrderedDict`...

Azad202222, 2023-07-24:
attribute_list = {nm_asr_node_attribute, nm_asr_node_identifier_attribute}
attribute_definitions = {ATTRIBUTE_DEFINITIONS['NmAsrNode'], ATTRIBUTE_DEFINITIONS['NmAsrNodeIdentifier']}
dbc_specifics = DbcSpecifics(attribute_list, attribute_definitions)
Now if we see the DbcSpecifics class:
class DbcSpecifics(object):

    def __init__(self,
                 attributes=None,
                 attribute_definitions=None,
                 environment_variables=None,
                 value_tables=None,
                 attributes_rel=None,
                 attribute_definitions_rel=None):
        if attributes is None:
            attributes = OrderedDict()

        if attribute_definitions is None:
            attribute_definitions = OrderedDict()
For None it is OrderedDict()?

Azad202222, 2023-07-24:
This is the error:
Traceback (most recent call last):
  File "D:\Python_Codes\main.py", line 200, in <module>
    dump_file(db, output_file)
  File "D:\Python_Codes\venv\lib\site-packages\cantools\database\__init__.py", line 248, in dump_file
    output = database.as_dbc_string(sort_signals=sort_signals)
  File "D:\Python_Codes\venv\lib\site-packages\cantools\database\can\database.py", line 340, in as_dbc_string
    return dbc.dump_string(InternalDatabase(self._messages,
  File "D:\Python_Codes\venv\lib\site-packages\cantools\database\can\formats\dbc.py", line 1845, in dump_string
    database = make_names_unique(database, shorten_long_names)
  File "D:\Python_Codes\venv\lib\site-packages\cantools\database\can\formats\dbc.py", line 1816, in make_names_unique
    make_node_names_unique(database, shorten_long_names)
  File "D:\Python_Codes\venv\lib\site-packages\cantools\database\can\formats\dbc.py", line 1747, in make_node_names_unique
    try_remove_attribute(node.dbc, 'SystemNodeLongSymbol')
  File "D:\Python_Codes\venv\lib\site-packages\cantools\database\can\formats\dbc.py", line 1737, in try_remove_attribute
    dbc.attributes.pop(name)
TypeError: set.pop() takes no arguments (1 given)