cantools/cantools #606: Screen flicker with  python -m cantools monitor -c can1 BigFile.arxml

robertrau, 2023-09-13:
First, thank you for all the work put in to cantools, it have served me well.
I am a hardware engineer writing a tester in python.
Running on a Raspberry Pi 3 Model B Rev 1.2
with a MCP2518 based HAT board
Using: Python 3.9.2
with a large (7.2MiB) arxml file
CAN bus: 500kbps, < 10% bandwidth
There is a lot of screen flicker that makes the screen hard to read.

Is there a terminal setting for curses that can reduce the flicker, or anything I can to do make the screen more readable?

Thank you.

robertrau, 2023-09-14:
My message loop is 1.1 seconds, so all messages have been received several times, See video.
Also, one of my CAN bus devices is getting power cycles, so there are some unacknowledged CAN messages. Is this a problem?

https://github.com/cantools/cantools/assets/18400100/0aac6010-de9f-46af-a770-83fee00b6540

andlaus, 2023-09-14:
I don't think that this is a problem related to anything that happens on the bus, but it's probably by how the `monitor` tool updates the screen. Unfortunately I cannot debug this myself because the issue does not crop up on any of my machines...

robertrau, 2023-09-18:
I found a fix from here:
https://stackoverflow.com/questions/24964940/python-curses-tty-screen-blink
I changed line 94 of monitor.py from clear to erase
and the screen is now flicker free.
Tested scrolling, paging up and down, messing with filters and the screen
stays up to date with no garbage left anywhere.

andlaus, 2023-09-20:
great! can you make a pull request?

robertrau, 2023-09-20:
OK, happy to, but full disclosure. I am a old hardware engineer and I have only used github for my own stuff (just pull, commit, and push). So I read the github tutorial at 
https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request#creating-the-pull-request
but got lost at step 2 where I should see my commits in a branch.
This is what I have done so far:
Created a work folder
Cloned the repo in the folder
edited the file with the change.

Please guide me through the next steps (sorry for needing this much hand holding).

andlaus, 2023-09-20:
okay, let's get this done. your next steps:

- create a [personal access token](https://docs.github.com/en/authentication/keeping-your-account-and-data-secure/managing-your-personal-access-tokens) (a classic one with the "repo" permissions is sufficient). keep this in a file so that you can easily copy-and-paste this later.
- fork the OG cantools repo to your personal github account (click the "Fork" button on https://github.com/cantools/cantools )
- create a remote for your personal fork in the local copy on your hard disk:
```
git remote add personal https://github.com/robertrau/cantools
git fetch personal
# (here you will be queried for the personal access token from above)
```
- create a feature branch for the PR
```
git checkout -b "remove_monitor_flicker"
```
- commit your changes to this branch
```
git commit -a
```
(make sure to write a meaningful commit message, i.e. it should confer *what* and *why* the changes are, not *how*)
- push the branch to the repository on your personal github account:
```
git push personal remove_monitor_flicker
```
(you will probably need to use your personal access token from above)
- visit https://github.com/cantools/cantools with your browser and click the green "Compare & pull request" button
- write a meaningful PR message (or re-use the one from the commit)

good look.

robertrau, 2023-10-15:
I couldn't understand the whole test system but I looks like you solved the issue another way (Mock?).
Thank you,

zariiii9003, 2023-10-19:
> I couldn't understand the whole test system but I looks like you solved the issue another way (Mock?). Thank you,

Yes, i just added a Mock 😄 

Closing, this should be solved in #609

closed 2023-10-19 by zariiii9003