cantools/cantools #610: The value for "message.cycle_time" behaves differently depending on the cantools version.

GilbertAhn, 2023-09-25:
In version 38.0.2, the cycle_time is output as defined in the DBC, but in version 39.2.0, messages without a cycle_time are output with a value other than None.

```
db = cantools.database.load_file('XXX.dbc')
for message in db.messages:
    # Tx message check
    if 'XXX' in message.senders:
        framesTx[message.name] = message
        txFrame.append(message.name)
        print(message.name, message.cycle_time)

    # Rx message check
    for signal in message.signals:
        if 'XXX' in signal.receivers:
            framesRx[message.name] = message
            rxFrame.append(message.name)
            print(message.name, message.cycle_time)
            break
```

In 38.0.2
```
XXX_200ms 200
XXX_200ms 200
XXX_FD None
XXX_GST None
XXX_00ms None
XXX_00ms None
XXX_00ms None
XXX_00ms None
XXX_00ms None
XXX_200ms 200
XXX_200ms 200
XXX_00ms None
XXX_00ms None
XXX_00ms None
XXX_00ms None
XXX_00ms None
XXX_00ms None
XXX_00ms None
XXX_00ms None
XXX_00ms None
XXX_00ms None
XXX_00ms None
XXX_FD 100
```

In 39.2.0
```
XXX_200ms 200
XXX_200ms 200
XXX_FD 200
XXX_GST 200
XXX_00ms 200
XXX_00ms 200
XXX_00ms 200
XXX_00ms 200
XXX_00ms 200
XXX_200ms 200
XXX_200ms 200
XXX_00ms 200
XXX_00ms 200
XXX_00ms 200
XXX_00ms 200
XXX_00ms 200
XXX_00ms 200
XXX_00ms 200
XXX_00ms 200
XXX_00ms 200
XXX_00ms 200
XXX_00ms 200
XXX_FD 100
```

GilbertAhn, 2023-09-26:
While it's good to take the default value when "GenMsgCycleTime" is 0, it seems appropriate to return "None" when the "send_type" is not "Cyclic".

```
    def cycle_time(self) -> Optional[int]:
        """The message cycle time, or ``None`` if unavailable.

        """

        if self._cycle_time is not None or not self.dbc:
            return self._cycle_time

        # fall back to default cycle time for DBC files
        try:
            if self._send_type is not None and self._send_type == "Cyclic":
                result = int(self.dbc.attribute_definitions['GenMsgCycleTime'].default_value)
                return None if result == 0 else result
            else:
                return None
        except (KeyError, TypeError):
            return None
```

zariiii9003, 2023-09-26:
This was changed in #592
@mon can you take a look?

mon, 2023-09-26:
Thanks for the @!

Unfortunately, basing behaviour around the value of `send_type` is impossible - almost every tool (including inside Vector's own documentation) uses a slightly different set of enum values - it's not standardised. I can make a list of all the ones I've seen, but it results in fragile code. Also, there are send types that still use GenMsgCycleTime that aren't cyclic: for example, `cyclicAndTriggered`, `cyclicIfActive`, `cyclicIfActiveAndTriggered`, among others.

If you would like the "raw" cycle_time, i.e. one that is `None` for any message that didn't define it, I recommend using the `_cycle_time` accessor.

Let me know if I misunderstood anything!

mon, 2023-09-26:
Although looking again, it is quite odd that you got `None` for default values in cantools 38 - the previous behaviour was to do exactly what the current code does, just at parse-time instead of access-time.
Could you possibly post your DBC file?

closed 2023-10-30 by andlaus

zariiii9003, 2023-11-01:
@GilbertAhn Could you try again with 39.3.0?

andlaus, 2023-11-02:
I restarted the workflow, and it seems to have worked: https://pypi.org/project/cantools/

PR #619 (2023-10-30): Refactor dbc GenMsgCycleTime handling
Closes #618 
Closes #610 (hopefully)