cantools/cantools #611: Mux values starting with Alpha fail to convert.

guy-radford-sunswap, 2023-09-26:
When setting a value for a Mux, if the value starts with an alpha, cantools, throws an error:
```
SYM: "Invalid syntax at line 3878, column 30: "Mux=FATAL_BAD_FAULT_TYPE 0,8 >>!<<C0h""
```

This version fails, the **Mux** line ends with the value `C0h`
```
[EDR_EVENT_FATAL]
ID=127h // Fatal event notification
Len=5
Mux=FATAL_BAD_FAULT_TYPE 0,8 C0h
Var=Context unsigned 8,32 // Fatal event context
```

If we change the Mux value to `90h` (starting with a number) the error is not thrown.
```
[EDR_EVENT_FATAL]
ID=127h // Fatal event notification
Len=5
Mux=FATAL_BAD_FAULT_TYPE 0,8 90h
Var=Context unsigned 8,32 // Fatal event context
```

swaldhoer, 2026-10-26:
Should this Not also be completed by #868?

juleq, 2026-10-26:
Yes, thanks. Closing.

closed 2026-10-26 by juleq