cantools/cantools #613: "division by zero" when the scale factor is 0

veerusajjan, 2023-10-04:
I am facing issue of "division by zero" when the scale factor is 0.
For example in the below:

***************************************************

BO_ 160 CLATIC_PANL: 8 DCD
 SG_ Rio_bt4 : 20|1@0+ (**0**,0) [0|1] "" ETM

***************************************************

The highlighted scale factor is 0, which is causing the division by zero error.
I am using cantools version "37.2.0"

Can anyone let us know what is the exact fix for this.?

[image]

Let us know your thoughts on this.

juleq, 2023-10-04:
Just tried, CANdb++ (which is considered authoritative for the dbc format) will not let you enter 0 as factor. As expected.

veerusajjan, 2023-10-05:
https://github.com/cantools/cantools/blob/356cc308de5a0470f221caf239f7560640527e23/src/cantools/database/conversion.py#L209

In the above line can we add the check for self.scale (I think here we are referring to the scaling factor) is zero. Is it preferable to have and avoid the divide by zero error.?

Let us know your thoughts on this.

juleq, 2023-10-05:
I would rather having it die in a fire than silently skip over errors in the dbc.

But I would not object to erroring out earlier (on dbc load).