cantools/cantools #616: CAN FD database

Kazaui, 2023-10-24:
Im trying to generate a database for can fd messages, when I open with dbc editor it shown the message as CAN Standard, also when is oppened in raw mode, no message type decalaration is made. May be I'm mistaking the command.

Here my code:
python
from cantools import database
new_db = database.Database()
new_signal = [database.can.Signal(name        = f"Signal_{x}", 
                                length      = 8, 
                                start       = x*8) for x in range (16)]
new_msg = database.can.Message(frame_id=100, name="MSG_0x64", length=32, signals=new_signal, is_fd=True)
new_db.messages.append(new_msg)
new_db.refresh()
database.dump_file(new_db, "FD_DBC.dbc")

Thank for advance

trulede, 2024-02-02:
Apparently you need to do this:

```
bus_type_def = AttributeDefinition("BusType",
                                           type_name="STRING",
                                           default_value="")
        db.dbc.attribute_definitions["BusType"] = bus_type_def
        db.dbc.attributes["BusType"] = Attribute(value="CAN FD",
                                                 definition=bus_type_def)
```

From: https://github.com/cantools/cantools/blob/cc02988affc529daa0fb0df042904ddbea1be02e/tests/test_convert.py#L139

ljq29, 2024-04-10:
Hi folks,

I'm encountering an issue while trying to create a DBC file for CAN FD from scratch, without using `cantools.db.load_file(fn_in)`. I've managed to successfully generate a DBC file for CAN 500, but when I attempt to add the following code to make the bus CAN FD:

```python
# make bus CAN FD
bus_type_def = AttributeDefinition("BusType",
                                   type_name="STRING",
                                   default_value="")
db.dbc.attribute_definitions["BusType"] = bus_type_def
db.dbc.attributes["BusType"] = Attribute(value="CAN FD", definition=bus_type_def)
```

I encounter the following error:

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[217], line 185
    181 # make bus CAN FD
    182 bus_type_def = AttributeDefinition("BusType",
    183                                    type_name="STRING",
    184                                    default_value="")
--> 185 db.dbc.attribute_definitions["BusType"] = bus_type_def
    186 db.dbc.attributes["BusType"] = Attribute(value="CAN FD", definition=bus_type_def)
    187 db.refresh()

AttributeError: 'NoneType' object has no attribute 'attribute_definitions'
```

Could anyone provide guidance or suggestions on how to resolve this issue? Any assistance would be greatly appreciated!

ljq29, 2024-04-10:
Here's the complete code I'm working with:

```python
db = cantools.database.can.Database(
        messages = messages,
        nodes = nodes,
        version='2.0'
    )

# make bus CAN FD
bus_type_def = AttributeDefinition("BusType",
                                   type_name="STRING",
                                   default_value="")
db.dbc.attribute_definitions["BusType"] = bus_type_def
db.dbc.attributes["BusType"] = Attribute(value="CAN FD", definition=bus_type_def)
db.refresh()

cantools.database.dump_file(db, dbc_filepath)
```

When I add the section to make the bus CAN FD, it leads to the error mentioned previously. I'm looking for advice on how to correctly implement this functionality without encountering the `AttributeError`. Any help or insight into what might be going wrong would be greatly appreciated.

ljq29, 2024-04-10:
> Apparently you need to do this:
>
> ```
> bus_type_def = AttributeDefinition("BusType",
>                                            type_name="STRING", [...]

Please help me answer this, thank you!

Kazaui, 2024-04-10:
Hi ljq29,

I dont test the script on creating new dbc from zero, and seams to be the issue, if you try to get the result of "db.dbc" is returns None, and this is the why u get the error.

Regards

ljq29, 2024-04-11:
So, can it be repaired?

brvcrypto, 2024-04-18:
> Please help me answer this, thank you!
       
       It works for me.
       cantools version: 39.3.0

        # Create a new database
        db = cantools.db.Database()
        
        #Create new dbc specifics
        dbc_spec = DbcSpecifics()

        # make bus CAN FD
        bus_type_def = AttributeDefinition("BusType",
                                           type_name="STRING",
                                           default_value="")
        dbc_spec.attribute_definitions["BusType"] = bus_type_def
        dbc_spec.attributes["BusType"] = Attribute(value="CAN FD", definition=bus_type_def)

        db.dbc = dbc_spec
        
        ...
        Fill dbc with content
        ...
        
        db.refresh()
        
        # Save the database to a DBC file
        cantools.db.dump_file(db, output_dbc_file)

Kazaui, 2024-10-01:
Hi @brvcrypto 

It works perfect, many thanks.

Just as extra, the import:
from cantools import database
from cantools.database.can.formats.dbc_specifics import DbcSpecifics
from cantools.database.can.attribute_definition import AttributeDefinition
from cantools.database.can.attribute import Attribute

regards,
Kazaui

closed 2024-10-01 by Kazaui