cantools/cantools #618: Example in the repo not working as expected

HARISH091089, 2023-10-27:
import os
import cantools
import pandas
import openpyxl

# Get the current directory
current_dir = os.getcwd()

SCRIPT_DIR = os.path.dirname(os.path.realpath(__file__))

INPUT_DBC_PATH = os.path.join(SCRIPT_DIR, 'XXXXXXXXX.dbc')
OUTPUT_DBC_PATH = os.path.join(SCRIPT_DIR, 'XXXXXXXXXX_new.dbc')

print("Loading DBC database from '{}'.".format(INPUT_DBC_PATH))
db = cantools.db.load_file(INPUT_DBC_PATH)
message = db.get_message_by_name('XXXXXXX')
try:
    print("Input cycle time: ",message.dbc.attributes['GenMsgCycleTime'].value)
    message.dbc.attributes['GenMsgCycleTime'].value = 2000
    print("Output cycle time:",message.dbc.attributes['GenMsgCycleTime'].value)
except KeyError as e:
    raise e
print("Input frame id: ", hex(message.frame_id))
message.frame_id = 0x19FF70EC
print("Output frame id:", hex(message.frame_id))
print("Writing modified DBC database to '{}'.".format(OUTPUT_DBC_PATH))
with open(OUTPUT_DBC_PATH, 'w', newline="\r\n") as fout:
    fout.write(db.as_dbc_string())

When I tried to run the above file, the GenMsgCycleTime is not getting affected instead it remains the same. Do you know as why this can happen?

closed 2023-10-30 by andlaus

HARISH091089, 2023-10-31:
Tried downloading the repo and updating the repo. I didn't see any upgrade of the software as the version seems to the same. 39.2.0 and it's the same now. Am I missing something here?
[image]

andlaus, 2023-10-31:
> Am I missing something here?

Yes: The patch was merged into the master branch, but no release was tagged, so the pypi package was not updated. I've tagged the latest master as v39.3.0, but the pypi package has unfortunately not been updated because we currently have an unrelated CI error in master. Any ideas how to manually upload the package @juleq or @zariiii9003?

zariiii9003, 2023-10-31:
> Yes: The patch was merged into the master branch, but no release was tagged, so the pypi package was not updated. I've tagged the latest master as v39.3.0, but the pypi package has unfortunately not [...]

We could remove the `needs` part and ALWAYS release on a tag:

````
  release:
    needs: [test, codegen, linter, build]
    runs-on: ubuntu-latest
    if: startsWith(github.ref, 'refs/tags')
```

Or we'll wait until the bitstruct fix is merged 😄

PR #619 (2023-10-30): Refactor dbc GenMsgCycleTime handling
Closes #618 
Closes #610 (hopefully)