cantools/cantools #625: TRC file support for plotting

joooeey, 2023-11-17:
I need to plot CAN data from TRC files in place of candump files.

There are (at least) two ways to go about this but none of them work out of the box:
- Add the ability to read TRC files directly into `cat data.trc | python -m cantools plot db.dbc`. TRC files are not supported in `cantools plot` right now. This requires a few new lines of code (basically two new regexes). I have this coded and tested locally. If there's interest from the maintainers to add this functionality, I'm happy to submit a PR including a unit test.
- Convert the TRC to candump with `python -m data.trc data.log` and then feed it into cantools: `cat data.log | python -m cantools plot db.dbc`. This doesn't work either because python-can adds an R at the end of every line in its candump (.log) and cantools doesn't know what to do with this R. I couldn't find any authoritative documentation on what that R is. In any case, python-can adds an R or a T in [this code line](https://github.com/hardbyte/python-can/blob/bc3f95544e7a7dbec0da855a0873c6ea502147fc/can/io/canutils.py#L183C29-L183C29) to indicate receipt or transmission. Perhaps we (I) can include this in the [`RE_CANDUMP_LOG` regex](https://github.com/cantools/cantools/blob/eb9e42d88a1e8dfde5d33cfff6a1fecfcdfbb531/src/cantools/subparsers/plot.py#L94) as well?

andlaus, 2023-11-17:
> Perhaps we (I) can include this in the [RE_CANDUMP_LOG regex](https://github.com/cantools/cantools/blob/eb9e42d88a1e8dfde5d33cfff6a1fecfcdfbb531/src/cantools/subparsers/plot.py#L94) as well?

we should probably go this route. I've never heard about the TRC file format, but if there alread is a way to convert it to the candump log format, let's leverage that. (PR welcome.)

joooeey, 2023-11-20:
> we should probably go this route. I've never heard about the TRC file format, but if there alread is a way to convert it to the candump log format, let's leverage that. (PR welcome.)

I'm still not sure if its wise to support what appears to be a non-standard format produced by python-can (i.e. `(1594172461.968006) vcan0 1F0#0000000000001BC1 R` with that ` R` that I haven't found in any other examples of candump files. Should cantools support only R and T at the end of the line? Or anything at all (i.e. remove the `$` from the regex)? Perhaps the change should even be upstream in python-can to add an option to not add that R?

joooeey, 2023-11-20:
I've added links to the original issue comment with what I would change. We'll test both as part of a larger system this week. And I'll see how this conversation goes. I intend to make a PR for one of the options or PRs for both options early in December.

andlaus, 2023-11-20:
> Should cantools support only R and T at the end of the line

As I see it, if the patch does not impede on any existing functionality and it is just a few lines of code, there are not many arguments against adding support for it.

If I'm not mistaken, changing the regular expression to
```
r'^\((?P<time>\d+\.\d+)\)\s+\S+\s+(?P<frameid>[\dA-F]+)#(?P<data>[\dA-F]*)( (?P<flags>[A-Z]*))?$'
```
should do the trick. note though, that the plot code does not (yet?) work with CAN-FD frames at all...

A better solution would be to refactor the plotting code to use [logreader.Parser](https://github.com/cantools/cantools/blob/master/src/cantools/logreader.py#L300). With this, the TRC file format *might* already be supported...

maybe @zariiii9003 has an opinion about this?

zariiii9003, 2023-11-20:
As one of the maintainers of python-can, i'd suggest to use the [`can.LogReader`](https://github.com/hardbyte/python-can/blob/bc3f95544e7a7dbec0da855a0873c6ea502147fc/can/io/player.py#L24) instead.
It's a dependency of cantools anyway.

Regarding the regex, i'll take a look at it later.

zariiii9003, 2023-11-20:
> I'm still not sure if its wise to support what appears to be a non-standard format produced by python-can (i.e. `(1594172461.968006) vcan0 1F0#0000000000001BC1 R` with that ` R` that I haven't found in any other examples of candump files. 

See [here](https://github.com/linux-can/can-utils/blob/41ea759237bbd156d753251c573c280d96873528/asc2log.c#L218)

joooeey, 2023-11-29:
> If I'm not mistaken, changing the regular expression to
> 
> ```
> r'^\((?P<time>\d+\.\d+)\)\s+\S+\s+(?P<frameid>[\dA-F]+)#(?P<data>[\dA-F]*)( (?P<flags>[A-Z]*))?$'
> ```
> 
> should do the trick. note though, that the plot code does not (yet?) work with CAN-FD frames at all.

What's the purpose of extracting the flags variable at this point?

joooeey, 2023-11-29:
> As one of the maintainers of python-can, i'd suggest to use the [`can.LogReader`](https://github.com/hardbyte/python-can/blob/bc3f95544e7a7dbec0da855a0873c6ea502147fc/can/io/player.py#L24) instead. It's a dependency of cantools anyway.

I just took a stab at this and gave up. The issue is that Cantools doesn't see the file suffix and `LogReader` can't guess the file type. It's not trivial to implement a reader that can guess all six formats supported by python-can since not all of them are linebreak-delimited.

I'm happy with the workaround mentioned in my second bullet point (converting with python-can and then feeding to cantools plot with support for the "R", as proposed in #630 ).

andlaus, 2023-12-05:
> It's not trivial to implement a reader that can guess all six formats supported by python-can since not all of them are linebreak-delimited.

maybe it would make sense to add something like the auto-detection capabilities the [cantools logreader](https://github.com/cantools/cantools/blob/master/src/cantools/logreader.py) into python-can, possible for only a subset of the formats supported by python-can?