cantools/cantools #629: Pythonic encode_message outputs invalid bytes

bash102, 2023-11-29:
When I attempt to encode a message from a loaded DBC with some signals above 31, the output bytearray is corrupted. It looks like the values are being interpreted as ascii. Here is the sample code using cantools 39.4.1
```
import cantools

# Load the DBC file
db = cantools.database.load_file(".\\Thunderstruck_TS2500_OBC.dbc")
#encode a message

message = db.encode_message('OBC_CMD', {'ConstantsA': 32, 'ConstantsB': 0, 'OBC_ControlCMD' : 0, 'OBC_LEDColor' : 0, 'OBC_MaxCurrentCMD' : 0, 'OBC_MaxVoltageCMD' : 0 })
print(message) #expecting b'\x80\x00\x00\x80\x0c\x00\x00\x00' got b'\x80\x00\x00\x80\x0c\x00\x00\x00'
message = db.encode_message('OBC_CMD', {'ConstantsA': 0, 'ConstantsB': 33, 'OBC_ControlCMD' : 0, 'OBC_LEDColor' : 0, 'OBC_MaxCurrentCMD' : 0, 'OBC_MaxVoltageCMD' : 0 })
print(message) #expecting b'\x00\x00\x00\x80\x0c\x00\x21\x00' got b'\x00\x00\x00\x80\x0c\x00!\x00'
```

I have seen this on a few dbc's I have now. I haven't been able to pin it down in the packing functions; the formatting of the signal seems right. So any suggestions would be appreciated!

The dbc in the example is attached with a modified extension so I could upload it.
[Thunderstruck_TS2500_OBC.dbc.txt](https://github.com/cantools/cantools/files/13496216/Thunderstruck_TS2500_OBC.dbc.txt)

bash102, 2023-12-02:
Looks like this is more of a printing issue with this data type. Does not affect functionality.

closed 2023-12-02 by bash102

andlaus, 2023-12-05:
note: to make these byte array objects readable by mere mortals, you can use their `.hex()` method...