cantools/cantools #633: Unexpected Behavior: Interesting Buffering Issue

markovejnovic, 2023-12-19:
I have observed that `grep` line-buffering is unexpected:

Let us run the following command:
```bash
stdbuf -oL candump can2 \
    | stdbuf -oL cantools decode -s "$MY_DBC_PATH" \
    | grep --line-buffered "$MY_MSG_NAME"
```

If the message `MY_MSG_NAME` is sent periodically, every one second, and no other message is present on the bus, you will observe that it takes a long time for this command to ever print out something and when it does it dumps a whole load of lines.

Contrast this with
```bash
stdbuf -oL candump can2 \
    | grep --line-buffered "$MY_MSG_ID"
```

and you will observe that every single time the message is received, it will be `grep`ed for and printed.

Setting:
```bash
export PYTHONUNBUFFERED=1
```

solved my issue, but I do believe it would be beneficial if the `--help` menu or the official documentation better documented this.

calumroy, 2024-11-12:
This was very helpful. Why is this not documented better?

closed 2026-10-23 by andlaus

markovejnovic, 2026-10-24:
Thank you, @timothyanderson096-ocdealcheck. I no longer use `cantools` (I'm out of embedded), but it puts a smile on my face to see issues I opened get closed and software improve =)