cantools/cantools #638: Why does the capitalization of signal names change

wangguodong-f, 2023-12-29:
In my dbc file, the signal name is capitalized, but in the generated c source file, the signal name becomes lowercase. why ?

andlaus, 2023-12-29:
this is basically a matter of taste: In C, camelCase is often frowned upon. Also, as I see it, the C source code generator should rather be considered as a template of how to implement such functionality and not as a one-size-fits-all ready-to-use tool. (On that matter I personally would warmly recommend using a template engine such as [`jinja`](https://jinja.palletsprojects.com) for implementing your own code generator instead of doing it the hard way like the shipped one.)

Also, if a given functionality is not of sufficient general interest, I am hesitant to accept patches adding new parameters for such things in order to avoid having a gazillion parameters complicating the generator for niche use cases...

timrulebosch, 2024-01-17:
Hi @andlaus, 

we are a bit interested :-) Our workflows would benefit from having some control on the emitted C code (i.e. not changing the variables at all). If there is a way we can achieve that, it would be very nice.

I opened an issue #644 and we are developing modification to achieve that. We would be happy to work on a PR.