cantools/cantools #641: A 2 byte Motorola signal that is signed is decoded wrong

funnel20, 2024-01-09:
We get a wrong value when decoding the raw can data.

Enclosed is a minified *.dbc file for 1 message only, and below you can find the simple steps to reproduce.
The problem is with this signal:
` SG_ signal_2 : 23|16@0- (0.1,0) [-300|600] "%" Vector__XXX`

To prove the issue, the following commands are used in a Terminal session on a MacBook with Python 3.9.7 installed:

- Install latest version of `cantools` lib: `python3 -m pip install cantools==39.4.2`
- Start a python session: `python3`
- Import lib: `import cantools`
- Add DBC file: `db = cantools.database.load_file('candb.dbc')`
- Decode the raw bytes, as has been observed with a CanKing from Krasser: `db.decode_message(42, bytes.fromhex("0000803200000000"))`

Overview of the above with the returned signal values:
```
>>> import cantools
>>> db = cantools.database.load_file('candb.dbc')
>>> db.decode_message(42, bytes.fromhex("0000803200000000"))
{'signal_1': 0.0, 'signal_2': -3271.8, 'signal_3': 0.65, 'signal_4': 0.65, 'signal_5': 'No Errors', 'signal_6': 'No Errors'}
```

The problem is with `signal_2` that returns a value of -3271.8 ❌

The raw 2 byte value 0x8032 should return -5.0, as explained:
The DBC specifies that it is a signed signal. So if the MSB of the raw data is 1, it is a negative value.
0x8032 in binary are:
[image]

The MSB (bit 15) is indeed high, so this is a negative value. 
The remaining 15 bits [image] have decimal value 50 (0x32).
According to the DBC file, there is a multiplier of 0.1 and 0 offset. Hence the value should be -50 *0.1 = **-5.0**  ⚠️
This has been also verified with CAN monitors, which show a value of -5.0 ✅

Why is the library showing value -3271.8 ❓

juleq, 2024-01-09:
Your explanation is wrong. Signed values are stored in 2s complement. If you consider this, the decode is correct.

juleq, 2024-01-09:
It is also intuitive that the result must be very negative as negative values closer to 0 would have more of their more significant bits set and not just the 0x80 bit.

funnel20, 2024-01-11:
@juleq Thank you for your prompt response. I have analysed this more and I must admit that you are correct. Thanks for pointing this out. I'm fine with closing this issue.

closed 2024-01-11 by juleq