cantools/cantools #652: Monitor doesn't work with Windows

felixFleetZero, 2024-03-22:
Tried to run cantools monitor tool on windows, running windows-curses v2.3.2, and it doesn't seem to work properly.
Running python 3.12.0, cantools 39.4.0.
The can adapter I'm trying to use (pcan-usb) works when using it with python-can only to receive raw CAN message from the same CAN bus.
So leads me to think it's an issue with the curses command line interface.
Note that I've also used the same dbc and can adapter on a raspberry pi with linux and managed to use cantools monitor successfully.

[image]

[image]

[image]

andlaus, 2024-03-25:
did you modify your local copy of cantools? the first three messages after invoking cantools seem to stem from something trying to load an excel file (which I'm pretty sure is not done by `cantools`)

That said, if I see this correctly, you did not specify any after `cantools monitor`, so maybe you just need to invoke tool correctly:
```
cantools monitor -c can0 my-bus-description-file.dbc
```

(be aware that the name of the can device on windows is most likely different, but I don't use cantools on windows, so I cannot tell you how it is usually called...)

felixFleetZero, 2024-03-25:
Sorry for the confusion, the second screenshot I shared was just to show the cantools version I had installed (first command in that screenshot, disregard the second).
See the first screenshot i shared, where i do indeed invoke the tool properly with a bus type, a channel, and a DBC file argument.
I haven't manually modified my local copy of cantools (I've got the same warnings when running ASAMMDF python module too, not sure if that could explains those warnings).
However, I also see those warnings on my Linux setup where cantools monitor actually works properly, so don't think they are related to the monitor issue.

felixFleetZero, 2025-11-20:
ahhh that's it, seems to work now! 
didn't know that option existed, did I miss it in the readthedoc?
thanks for the tip 👌