cantools/cantools #664: Extract Request and Response CAN IDs from CDD File

Janelyol, 2024-05-17:
In addition to all the DIDs (Diagnostic Identifiers), I require the Request CAN ID and the Response CAN ID from the Candela file (CDD). These CAN IDs will serve as the default communication IDs for interacting with the ECU.
Typically, these IDs are readily available in the CDD file and can be extracted by parsing the qualification names `CAN.ReqCanId` and `CAN.ResCanId`.

andlaus, 2024-05-20:
support for the diagnostic functionality is severely limited in cantools by design. I recommend using [odxtools](https://github.com/mercedes-benz/odxtools) as a fairly emcompassing alternative. (you need to export your CDD files to PDX using CANdela's ODXstudio plugin.)

andlaus, 2024-05-20:
BTW: If I understood you correctly, your use case is already answered in odxtool's [README](https://github.com/mercedes-benz/odxtools?tab=readme-ov-file#python-snippets)...