cantools/cantools #665: For Autosar classic platform release R4.2.0 and R4.2.1, I am not getting the schema version of it. Can anyone help?

VarshaVrush, 2024-05-22:
There are specific schema version for each classic platform versions for example:
Classic platform release       Schema versions
 R4.3.0                                    AUTOSAR_00042

In the same way, I am not getting schema version of R4.2.0

VarshaVrush, 2024-05-22:
this is the refernce for the issue

andlaus, 2024-05-22:
as far as I am aware, for the functionality relevant to cantools there is basically no difference between any of the AUTOSAR 4.x (and newer) releases. It seems that AUTOSAR redesigned their website which lead to them currently not providing the documents for AUTOSAR classic versions older than 4.3.1. But as said, any of the versions which they currently still provide should be fine regarding all cantools functionality.