cantools/cantools #667: Error while trying to run 'monitor' subparser from CLI

jiyojolly, 2024-06-10:
Hi,
I am getting the following error while trying to run 'monitor' subparser from CLI. 
Error:  "error: Can't instantiate abstract class Monitor with abstract method stop*

Cantools Version: 39.4.5
OS: Ubuntu 22.04.04 

It is a fresh install of Ubuntu. I have tried reinstalling cantools, pip, purging pip cache etc. I can confirm other subparses like list, decode, dump etc. is working as expected. 

Please let me know if you require any more information.

jiyojolly, 2024-06-11:
I was able to pinpoint the issue to the python-can package dependency. The latest version, python-can 4.4.0, causes an error. I rolled back the package to version 4.3.0, and that fixed the problem. It seems that the definition of the can.Listener base class, from which the derived Monitor class inherits, has changed.

mendola, 2024-06-12:
Here's a possible workaround for preventing that error while using python-can 4.4.0: https://github.com/cantools/cantools/pull/668