cantools/cantools #670: Incorrect Float parsing in DBC files

AltayAkkus, 2024-06-30:
Hello,
really appreciate your work but I encountered a small problem:

## Steps to reproduce:
```
>>> import cantools
>>> db = cantools.db.load_file('vw_mqb_2010.dbc')
>>> signal = db.messages[0].signals[8]
>>> signal
signal('ACC_Sollbeschleunigung_02', 24, 11, 'little_endian', False, None, 0.005, -7.22, -7.22, 3.005, 'Unit_MeterPerSeconSquar', False, None, {2046: 'Neutralwert', 2047: 'Fehler'}, None, None)
>>> signal.is_float
False
```

The dbc file is from the [OpenDBC repo](https://github.com/commaai/opendbc/blob/master/vw_mqb_2010.dbc)

The signal seems pretty `float`y to me, see the `minimum` and `maximum` in the example, but `is_float` return `False`.

I've looked in the source code, and this seems to be the appropriate function which sets this property
https://github.com/cantools/cantools/blob/2c92c60f2d1b50894ad0c5652afd331283e03402/src/cantools/database/can/formats/dbc.py#L1470-L1478

It apparently searches for references in the DBC file which indicate whether this is a float, a double or whatever, but this file doesn't have attributes with `SIG_VALTYPE`

https://github.com/cantools/cantools/blob/2c92c60f2d1b50894ad0c5652afd331283e03402/src/cantools/database/can/formats/dbc.py#L1304-L1316

The file is weird in general, but parsed by some software, apparently. Like, look at the 11 bit signed float? 

The question one may ask: Could we just derive the data type from `type(signal.minimum)`? Or has someone else encountered this weird stuff?
Is this maybe an integer mapped to a float (internally), and the minimum maximum here are just annotations? Should we treat minimum and maximum as annotations, since some programs/generators/editors do that??

Happy to hear your thoughts.

juleq, 2024-06-30:
All good. That is an 11 bit integer on the wire (raw). CANdb distinguishes between raw and physical. Raw is the value on the wire (can be signed or unsigned int of arbitrary length or, in rare cases, IEEE 754 floats/doubles). Physical is the real-world value after scale and offset have been applied to it. So there is a simple linear transformation between raw and physical. And this transformation is done in some float type where that is more convenient (some embedded implementations rather use a fix-point internal representation for performance reasons).

Min and max are given in the physical domain.

juleq, 2024-06-30:
Raw = (Physical - Offset) / Scale would, for max, be:

2045 = (3.005 - (-7.22)) / 0.005

That is suspiciously close to full scale 2^11 = 2048. So the transformation spreads it out such that it can be transferred as integer. With acceptable resolution of 0.005 m/s^2.

hongyeee, 2024-08-05:
I asked same question few weeks ago.

https://github.com/cantools/cantools/issues/673

This issue can help you!