cantools/cantools #672: Incorrect ID conversion when converting from SYM to DBC

mupfdev, 2024-07-04:
Hello,
I have converted a large SYM database to DBC. Some high IDs have been converted incorrectly.

Example:

`18FF15F3h` ended up as `98FF15F3h` (2566854131)

**0**0011000111111110001010111110011
**1**0011000111111110001010111110011

Under certain circumstances, the uppermost bit is set, although this should not be the case.

Best regards,
Michael

juleq, 2024-07-04:
Hi.

The dbc file format uses the most significant bit of a 32 bit id word to indicate the id to be either extended (29 bits wide) or standard (11 bits).

Other file formats have an atomic attribute to indicate extendedness...

mupfdev, 2024-07-04:
Oh, thank you. I did not know that. I'll fix my parser!

closed 2024-07-04 by mupfdev