cantools/cantools #673: Signal.is_float doesn't change from False to True

hongyeee, 2024-07-05:
Hello.

Great work with this library:)

I encountered a problem using `signal.is_float`. 

I would expect to be able to write the following:

``` sql
SG_ Roll_Rate : 0|16@1+ (0.048831105,-163.84) [-163.84|163.835] "m/s"  ABS
```
This is a line from the normal dbc file.

I expect (Roll_Rate) `signal.is_float` is `True`.

However, 

```
>>> signal
signal('Roll_Rate', 0, 16, 'little_endian', False, None, 0.048831105, -163.84, -163.84, 163.835, 'm/s', False, None, None, None, {None: 'Measured roll rate around the X axle.'})
>>> signal.is_float
False
```
The signal seems `float`, seeing the `scale`, `offset`, `minimum` and `maximum` in the example, but `is_float` returns `False`.

I've looked in the source code, and this seems to be the inappropriate function.

https://github.com/cantools/cantools/blob/2c92c60f2d1b50894ad0c5652afd331283e03402/src/cantools/database/conversion.py#L50-L65

https://github.com/cantools/cantools/blob/2c92c60f2d1b50894ad0c5652afd331283e03402/src/cantools/database/conversion.py#L186-L220

According to `LinearConversion` class, it get scale and offset by parameter.(Their data types are explicitly `float`.)
But, `self.is_float` doesn't change to `True`.

Is this right way to indicate whether the raw value is a float or not?

Thanks a lot for your help.

zariiii9003, 2024-07-05:
If the raw value was a float, you'd have something like "SIG_VALTYPE_ 123 Roll_Rate : 1;" in your dbc file.

hongyeee, 2024-07-07:
I can't find SIG_VALTYPE_ symbol used in my dbc file. 
It means the raw value wasn't a float type.
However, the physical value can be a float because of `scale` and `offset`. 

Do I understand correctly?

andlaus, 2024-07-08:
yes. somewhat confusingly, the `Signal.is_float` attribute means that the raw value is floating point, not the physical one. That#s because the latter conceptually is always floating point for numerical signals.

hongyeee, 2024-07-11:
I got it! Thanks you so much:)

closed 2024-07-15 by hongyeee