cantools/cantools #680: Failed to set choices for cdd file

swatabc, 2024-08-06:
Hello, there was an issue while using version 39.4.5.

In my cdd file, the following content is ok:
~~~xml
<TEXTMAP s='0' e='0'>
~~~
But this scenario is an error:
~~~xml
<TEXTMAP s='(0,0,0,0,31,127,182,60,0,0,174,1,255,38,0,12,18,6,220,20,8,30,214,0)' e='(0,0,0,0,31,127,182,60,0,0,174,1,255,38,0,12,18,6,220,20,8,30,214,0)'>
~~~

I have checked the source code and it is the following logic that caused the error:
~~~python
for choice in data_type.findall('TEXTMAP'):
        start = int(choice.attrib['s'].strip('()'))
        end = int(choice.attrib['e'].strip('()'))
~~~

May I ask if my file is not standard or if the program is not compatible.

Thank you.

andlaus, 2024-08-06:
cantools is quite strongly aimed at regular signal based communication, not diagnostics (nowadays I consider the diagnostics code in cantools as technical debt a  which cannot be easily removed due to the trouble it would cause by breaking user code). If you want to do some serious diagnosing, I would strongly advice to convert your CDD file to ODX using the ODX-studio plugin for CANdela and use the resulting file with [odxtools](https://github.com/mercedes-benz/odxtools)

swatabc, 2024-08-19:
> cantools 主要是针对常规信号通信，而不是诊断（如今我认为 cantools 中的诊断代码是一种技术债务，由于破坏用户代码会造成麻烦，因此无法轻易删除）。如果您想进行一些严肃的诊断，我强烈建议使用 CANdela 的 ODX-studio 插件将您的 CDD 文件转换为 ODX，然后将生成的文件与 odxtools 一起[使用](https://github.com/mercedes-benz/odxtools)

Thank you for your suggestion.

closed 2024-08-19 by andlaus