cantools/cantools #686: Truncated signal names to 32 characters in a signal_group

AberciF, 2024-08-20:
Hello, I found a bug or a problem that I can't deal with.
Why are signal names truncated to 32 characters in signal_groups ?
Signal names in a signal_group are not linked to the right name (long if it's more than 32 characters)
Due to that fact, it's impossible to retreive (link) truncated signal name to the right (long) signal name.

- For instance, I have a DBC with the following attributes:
```...
BA_ "SystemSignalLongSymbol" SG_ 2583690946 AAA_actCcsControlPilotState_X_CC "AAA_actCcsControlPilotState_X_CC_EP2";
BA_ "SystemSignalLongSymbol" SG_ 2583690946 AAA_actCcsConnectorStatus_X_CC_S "AAA_actCcsConnectorStatus_X_CC_SqC_EP2";
BA_ "SystemSignalLongSymbol" SG_ 2583690946 AAA_actCcsConnectorStatus_X_CC_E "AAA_actCcsConnectorStatus_X_CC_EP2";
BA_ "SystemSignalLongSymbol" SG_ 2583690946 AAA_actCcsConnectorStatus_X_CC_C "AAA_actCcsConnectorStatus_X_CC_CRC_EP2";
...
SIG_GROUP_ 2583690946 AAA_actCcsConnectorStatus_C1_sg 1 : AAA_actCcsInletLock_X_CC_EP2 AAA_actCcsControlPilotState_X_CC AAA_actCcsConnectorStatus_X_CC_S AAA_actCcsConnectorStatus_X_CC_E AAA_actCcsConnectorStatus_X_CC_C;
```

- If I call this code:
```
db = cantools.database.load_file(dbc_path)
for x in db.messages:
    for sg_names in x.signal_groups:
        print(sg_names)
        
        for signal_name in sg_names.signal_names:
            print(signal_name)
```

- I have this result:

> signal_group('AAA_actCcsConnectorStatus_C1_sg', 1, ['AAA_actCcsInletLock_X_CC_EP2', 'AAA_actCcsControlPilotState_X_CC', 'AAA_actCcsConnectorStatus_X_CC_S', 'AAA_actCcsConnectorStatus_X_CC_E', 'AAA_actCcsConnectorStatus_X_CC_C'])
> 
> AAA_actCcsInletLock_X_CC_EP2
> AAA_actCcsControlPilotState_X_CC
> AAA_actCcsConnectorStatus_X_CC_S
> AAA_actCcsConnectorStatus_X_CC_E
> AAA_actCcsConnectorStatus_X_CC_C

- In this case I expected the result to be like that:

> signal_group('AAA_actCcsConnectorStatus_C1_sg', 1, ['AAA_actCcsInletLock_X_CC_EP2', 'AAA_actCcsControlPilotState_X_CC_EP2', 'AAA_actCcsConnectorStatus_X_CC_SqC_EP2', 'AAA_actCcsConnectorStatus_X_CC_EP2', 'AAA_actCcsConnectorStatus_X_CC_CRC_EP2'])
> 
> AAA_actCcsInletLock_X_CC_EP2
> AAA_actCcsControlPilotState_X_CC_EP2
> AAA_actCcsConnectorStatus_X_CC_SqC_EP2
> AAA_actCcsConnectorStatus_X_CC_EP2
> AAA_actCcsConnectorStatus_X_CC_CRC_EP2

AberciF, 2024-11-25:
Nope ! and I haven't any response :-(

zariiii9003, 2024-11-25:
I'm not sure, what the question here is. Cantools always uses short names. You can access the long name via `signal.dbc.attributes['SystemSignalLongSymbol'].value`

AberciF, 2024-11-25:
Hello, It's not a question it's a fact !
Signal names are truncated to 32 characters in signal_groups.
In this case we are not able to link signals inside a signal_groups.

zariiii9003, 2024-11-26:
> Hello, It's not a question it's a fact ! Signal names are truncated to 32 characters in signal_groups. In this case we are not able to link signals inside a signal_groups from the signal list.

Could you upload your dbc file?

realmaryam, 2025-01-26:
I have the same issue. e.g. I want to access the signal names of a group in a message, but the name list in signal group does not contain the total name of signal, but only contains 32 character.
`
for message in db.messages:
            if message.signal_groups:
                     signals = [message.get_signal_by_name(signal_name) for signal_name in message.signal_groups[0].signal_names]
`
issue:
`
return self._signal_dict[name]
           ~~~~~~~~~~~~~~~~~^^^^^^
KeyError: 'EPSAngleCtrlDlvdTorsionBarTorque'
`

In this case, the complete name of the signal is  'EPSAngleCtrlDlvdTorsionBarTorqueValue_S'

zariiii9003, 2025-01-26:
@realmaryam That should be fixed in #713

realmaryam, 2025-01-27:
> [@realmaryam](https://github.com/realmaryam) That should be fixed in [#713](https://github.com/cantools/cantools/pull/713)

Thanks, that was really helpful.

AberciF, 2025-01-28:
> @realmaryam That should be fixed in #713 

Thanks, I will test that.

AberciF, 2025-04-24:
I confirm, it's solved.
Thanks, for your great job !

closed 2025-04-24 by AberciF